PubInv/NASA-COG #219: UPS for DUE Independent of The Rest of the Loads

ForrestErickson, 2023-09-08:
If building power is lost, the DUE needs some uninterruptable power supply independent of building power.

LokiMetaSmith, 2023-09-08:
I would recommend either a USB power pack or a separate 4.7v lithium pack

I did a quick test with a USB power bank and it did indeed work as a ups,
however it temporarily disabled DC voltage as it switched over. This can
either be accepted in software or a sufficiently large ride through
capacitor can be used to keep it running

This doesn't solve the issue with relying on Internet for data capture if
power really does go down

LokiMetaSmith, 2023-09-09:
During a call with Robert, an idea was presented. 

It is the case that a 12v SLA( Sealed Lead Acid) battery has the property of being able to be float charged to 12-13v indefinitely, which means that as a temporary solution we can parallel a 12v battery without risk of damage to the meanwell power supply (It's got a reverse bias protection diode built in) or the Due (which is the only other DC powered circuit in the system currently), which has the property in which when the DC power supply looses power, the SLA can take over providing power to the system. 

It has been observed that the system "pulls less than an amp during normal operations" which means that a 4-10 Amp Hour SLA battery should last at a minimum 4-10 hours, which is more than enough to effect a safe shutdown.

This is envisioned as a stop gap modification of ODECS 1# and 2#

LokiMetaSmith, 2023-09-09:
this has been implemented for ODECS #2

ForrestErickson, 2023-09-09:
Regarding, "this has been implemented for ODECS https://github.com/PubInv/NASA-COG/issues/2"
Make reproducible with photos and wiring description please.

How might I approximate this on my Due and Ethernet setup for example?

ForrestErickson, 2023-09-09:
Regarding, "This is envisioned as a stop gap modification of ODECS 1# and 2#"

I have changed the issue name to make clear the scope of this fix.

LokiMetaSmith, 2023-10-06:
Between your 12v power supply and the DUE, put in parallel a SLA battery rated for 12v. Then adjust your power supply for 12.7v as that is the constant dc float voltage that is allowed on SLA batteries.

closed 2023-10-06 by LokiMetaSmith