PubInv/NASA-COG #222: Indecision on J11, J12 and J13 Connector or Screw Terminal

ForrestErickson, 2023-09-09:
# Indecision on J11, J12 and J13 Connector or Screw Terminal
[image]

We have conflicting solutions for  J11, J12 and J13

### Connectors as build (JLCPCB or Lawrence?)
[image]
What is the part number on this part as built?  

### Lee put screw terminals in the schematic on or about 20230830
[image]
The schematic has JLCPCB part number C918120

## Which do we want?
Is the PCB foot print appropriate?
Is pin 1 properly defined?
Should GRN, and Voltage and polarity be marked on the PCB ?

LokiMetaSmith, 2023-09-09:
For prototyping, screw terminals are preferred, for production, crimped connectors are preferred. The choice of crimped connectors was based on a bom error and the need to "get it done ASAP" and so errors were introduced. In this case the error is an opportunity to discuss the merit of the chosen interface. A screw terminal has the unfortunate problem of being difficult to verify the quality of the connection being made, where as a crimp can be visually inspected and is a more repeatable process. It also allows for a diversity of crimp types and current handling capabilities. 

- Is the PCB foot print appropriate? 
- -2.54mm is a bit low for anything exceeding 12v, the standard spacing is 5.08mm or 5mm depending on if it's imperial or metric
- Is pin 1 properly defined?
- - Pin 1 should be in most cases the power pin or the positive pin if possible, with the other pin being the negative or ground pin, but there is no official standard
- Should GRN, and Voltage and polarity be marked on the PCB ?
- - Yes, the silk screen regardless of which is chosen needs more work to improve legibility.

ForrestErickson, 2023-10-06:
@LokiMetaSmith 
@RobertLRead 

Regarding, "In this case the error is an opportunity to discuss the merit of the chosen interface. "
Lee too has mixed feeling about screw terminals versus a crimped terminal.  A good screw terminal connection will be better than a clumsy hand / pliers smashed crimp connection which is the only kind Lee can make at present.

Neither Robert nor Lee have crimping tools. If we are to move to crimped connections we should standardized on a connector type and tools and purchase tools for all three locations as well as a supply of terminals and shells and headers. 

Another solution is to purchase preterminated cable assemblies and cut them but this will be a frequent fishing expedition. 

Lawrence, Remind us again of the crimp tool and terminals you use please. I seam to remember considering purchasing a kit of some stuff from Amazon but never followed through.

LokiMetaSmith, 2023-10-06:
> If we order crimping tools and materials for Lee and Robert then we will
> close this having decided to move to headers.
>
> If we choose to to with screw terminals perhaps we should use the same pin
> pitch as on the +12 and +24 supplies. What are those spacings?

Iwiss XH2.0mm XH2.54mm XH3.96mm Dupont D-Sub Terminals JST Pin Crimper
SN-01BM Crimping Tools for Crimper Plier for AWG 28-20 （0.08-0.5mm2）for
power connectors
https://www.amazon.com/gp/product/B00YGLKBSK/

Micro Connector Pin Crimping Tool 0.03-0.52mm² 32-20AWG Ratcheting Crimper
for D-Sub,Open Barrel suits Molex,JST,JAE for power supply crimps
https://www.amazon.com/gp/product/B078WPT5M1/

ferrule hex crimp for loose wires and screw terminal connections
https://www.amazon.com/gp/product/B00H950AK4/

ForrestErickson, 2023-10-12:
@LokiMetaSmith 

Currently the Schematic still has screw terminals with a JLCPCB part number. If we order this way we get screw terminals.
Please provide to me the exact JLCPCB part number for the header which you want to use for locations below.

[image]

LokiMetaSmith, 2023-10-12:
the branch develop has the updated part number C265283

https://jlcpcb.com/partdetail/Jst_SalesAmerica-B2B_XH_AGU/C265283

LokiMetaSmith, 2023-10-12:
crimp header kit
https://www.amazon.com/gp/product/B09DBGVX5C/

ForrestErickson, 2023-10-12:
@LokiMetaSmith 
What about a part number for the terminal for use with stranded wire into screw terminals?