PubInv/NASA-COG #225: Due, Ethernet Running, Halts. Log Files

ForrestErickson, 2023-09-09:
## The Due with ethernet running in Lee's lab halts after about quarter hour.

### Software under test is from develop branch commit: 
[74941c8](https://github.com/PubInv/NASA-COG/commit/74941c834f0ffa662fdd48febc9b65bd9bd144be)

### Hardware description

1. Due
2. Seeeds Studio W52000 ethernet shield.
3. Network connection.
4. Power from USB hub.
5. 

### Results

* The file log2_20230909_1838.txt halted after on the order of 14 minuets. The exact time is unknown because the capture started after the reset.

* The file Log3_20230909_1902.txt halted after about 15 minuets. The full run time was captured as the log was started while the Due was held in reset.

Both log files are in this single zip folder.
[Log3_20230909_1902.zip](https://github.com/PubInv/NASA-COG/files/12566892/Log3_20230909_1902.zip)

### Notes on Halt

The last burst of data ends the same way on both haul events

Halt of log2_20230909_1838.txt

> [2023-09-09 22:38:01.481] Machine State: Off
> [2023-09-09 22:38:01.481] Target      C: 30.00
> [2023-09-09 22:38:01.485] Post Heater C: 0.00
> [2023-09-09 22:38:01.485] Post Getter C: 0.00
> [2023-09-09 22:38:01.485] Post Stack  C: 0.00
> [2023-09-09 22:38:01.489] Stack volts V: 0.00
> [2023-09-09 22:38:01.489] Stack amps  A: 0.00
> [2023-09-09 22:38:01.493] Stack ohms  O:  N/A
> [2023-09-09 22:38:01.493] Heater DC    : 0.0000
> [2023-09-09 22:38:01.497] Fan PWM      : 0.00
> [2023-09-09 22:38:01.497] Fan RPM      : 0.00
> [2023-09-09 22:38:04.409] Error: Could not read temperature data: 0
> [2023-09-09 22:38:04.413] Error: Could not read temperature data: 0
> [2023-09-09 22:38:04.418] Error: Could not read temperature data: 0
> [2023-09-09 22:38:04.423] Error: Could not read temperature data: 1
> [2023-09-09 22:38:04.426] Error: Could not read temperature data: 2
> [2023-09-09 22:38:04.430] Bad post_heater_C
> [2023-09-09 22:38:04.430] Bad post_getter_C
> [2023-09-09 22:38:04.430] Bad post_stack_C

Halt of log3_20230909_1902.txt

> [2023-09-09 23:17:55.849] Machine State: Off
> [2023-09-09 23:17:55.853] Target      C: 30.00
> [2023-09-09 23:17:55.853] Post Heater C: 0.00
> [2023-09-09 23:17:55.857] Post Getter C: 0.00
> [2023-09-09 23:17:55.857] Post Stack  C: 0.00
> [2023-09-09 23:17:55.857] Stack volts V: 0.00
> [2023-09-09 23:17:55.861] Stack amps  A: 0.00
> [2023-09-09 23:17:55.861] Stack ohms  O:  N/A
> [2023-09-09 23:17:55.866] Heater DC    : 0.0000
> [2023-09-09 23:17:55.866] Fan PWM      : 0.00
> [2023-09-09 23:17:55.869] Fan RPM      : 0.00
> [2023-09-09 23:17:58.782] Error: Could not read temperature data: 0
> [2023-09-09 23:17:58.786] Error: Could not read temperature data: 0
> [2023-09-09 23:17:58.790] Error: Could not read temperature data: 0
> [2023-09-09 23:17:58.794] Error: Could not read temperature data: 1
> [2023-09-09 23:17:58.794] Error: Could not read temperature data: 2
> [2023-09-09 23:17:58.799] Bad post_heater_C
> [2023-09-09 23:17:58.802] Bad post_getter_C
> [2023-09-09 23:17:58.802] Bad post_stack_C

ForrestErickson, 2023-09-10:
Reviewed the above with Robert and decided to make experiment with the ethernet shield removed and log the same test. 
Further bulletins as events warrant!

ForrestErickson, 2023-09-10:
### Test with Out Ethernet Shield
I made two runs with the ethernet shield removed.

 Both  terminated  quickly.

* File: Log4_20230909_2005.txt
> [2023-09-10 00:05:51.751] Starting Ox...
> [2023-09-10 00:05:51.849] Create watchdog (todo)
> [2023-09-10 00:05:51.853] Core booted...
> [2023-09-10 00:05:51.952] BEGINNING ASSERTION CHECKS!!
> [2023-09-10 00:05:51.952] IF YOU DO NOT SEE THE WORDS 'ALL CLEAR' BELOW AN ASSERTION HAS FAILED
> [2023-09-10 00:05:52.049] ALL CLEAR!!
> [2023-09-10 00:05:53.102] didn't set address
> [2023-09-10 00:05:54.152] didn't set address
> [2023-09-10 00:05:54.152] Manuf: UNKWN
> [2023-09-10 00:05:55.699] didn't set address
> [2023-09-10 00:05:55.703] Model: 
> [2023-09-10 00:05:58.403] didn't set address
> [2023-09-10 00:05:58.403] VoltageSt: 
> [2023-09-10 00:05:58.801] didn't set address
> [2023-09-10 00:05:58.804] Rev: 
> [2023-09-10 00:06:00.352] didn't set address
> [2023-09-10 00:06:00.352] ManufDate: 
> [2023-09-10 00:06:01.901] didn't set address
> [2023-09-10 00:06:01.905] Serial: 
> [2023-09-10 00:06:03.453] didn't set address
> [2023-09-10 00:06:03.453] Country: 
> [2023-09-10 00:06:05.001] didn't set address
> [2023-09-10 00:06:05.001] RateVoltage: 65535
> [2023-09-10 00:06:06.551] didn't set address
> [2023-09-10 00:06:06.554] RateCurrent: 65535
> [2023-09-10 00:06:07.049] MaxVoltage: 65535
> [2023-09-10 00:06:07.554] MaxCurrent: 65535
> [2023-09-10 00:06:09.102] didn't set address
> [2023-09-10 00:06:09.102] failed to turn it on
> [2023-09-10 00:06:10.150] didn't set address
> [2023-09-10 00:06:10.154] failed to set volts
> [2023-09-10 00:06:11.204] didn't set address
> [2023-09-10 00:06:11.204] failed to set current
> [2023-09-10 00:06:11.207] ReadTempsTask init
> [2023-09-10 00:06:11.207] Locating devices...Found 0 devices.
> [2023-09-10 00:06:11.212] Parasite power is: OFF
> [2023-09-10 00:06:11.212] Unable to find address for Device 0
> [2023-09-10 00:06:11.216] Unable to find address for Device 1
> [2023-09-10 00:06:11.220] Unable to find address for Device 2
> [2023-09-10 00:06:11.224] Device 0 Address: 0000000000000000
> [2023-09-10 00:06:11.224] Device 1 Address: 0000000000000000
> [2023-09-10 00:06:11.228] Device 2 Address: 0000000000000000
> [2023-09-10 00:06:11.232] Device 0 Resolution: 0
> [2023-09-10 00:06:11.232] Device 1 Resolution: 0
> [2023-09-10 00:06:11.236] Device 2 Resolution: 0
> [2023-09-10 00:06:11.236] Config of temperature sensors done
> [2023-09-10 00:06:11.240] CogTask init
> [2023-09-10 00:06:11.240] TempRefreshTask init
> [2023-09-10 00:06:11.244] starting
> [2023-09-10 00:06:11.854] The MAC address is: BBB
> [2023-09-10 00:06:12.854] AAA
> [2023-09-10 00:06:12.854] QQQQ
> [2023-09-10 00:06:12.854] ZZZZZZ
> [2023-09-10 00:06:12.854] MMMMMMM
> 

* File:  Log5_20230909_2039.txt

> [2023-09-10 00:39:14.368] Starting Ox...
> [2023-09-10 00:39:14.471] Create watchdog (todo)
> [2023-09-10 00:39:14.471] Core booted...
> [2023-09-10 00:39:14.569] BEGINNING ASSERTION CHECKS!!
> [2023-09-10 00:39:14.573] IF YOU DO NOT SEE THE WORDS 'ALL CLEAR' BELOW AN ASSERTION HAS FAILED
> [2023-09-10 00:39:14.667] ALL CLEAR!!
> [2023-09-10 00:39:15.719] didn't set address
> [2023-09-10 00:39:16.768] didn't set address
> [2023-09-10 00:39:16.772] Manuf: UNKWN
> [2023-09-10 00:39:18.321] didn't set address
> [2023-09-10 00:39:18.321] Model: 
> [2023-09-10 00:39:19.870] didn't set address
> [2023-09-10 00:39:19.870] VoltageSt: 
> [2023-09-10 00:39:21.417] didn't set address
> [2023-09-10 00:39:21.422] Rev: 
> [2023-09-10 00:39:22.971] didn't set address
> [2023-09-10 00:39:22.971] ManufDate: 
> [2023-09-10 00:39:24.518] didn't set address
> [2023-09-10 00:39:24.521] Serial: 
> [2023-09-10 00:39:26.071] didn't set address
> [2023-09-10 00:39:26.071] Country: 
> [2023-09-10 00:39:27.620] didn't set address
> [2023-09-10 00:39:27.623] RateVoltage: 65535
> [2023-09-10 00:39:29.171] didn't set address
> [2023-09-10 00:39:29.171] RateCurrent: 65535
> [2023-09-10 00:39:29.671] MaxVoltage: 65535
> [2023-09-10 00:39:30.170] MaxCurrent: 65535
> [2023-09-10 00:39:31.720] didn't set address
> [2023-09-10 00:39:31.723] failed to turn it on
> [2023-09-10 00:39:32.767] didn't set address
> [2023-09-10 00:39:32.772] failed to set volts
> [2023-09-10 00:39:33.821] didn't set address
> [2023-09-10 00:39:33.821] failed to set current
> [2023-09-10 00:39:33.825] ReadTempsTask init
> [2023-09-10 00:39:33.825] Locating devices...Found 0 devices.
> [2023-09-10 00:39:33.829] Parasite power is: OFF
> [2023-09-10 00:39:33.829] Unable to find address for Device 0
> [2023-09-10 00:39:33.833] Unable to find address for Device 1
> [2023-09-10 00:39:33.836] Unable to find address for Device 2
> [2023-09-10 00:39:33.841] Device 0 Address: 0000000000000000
> [2023-09-10 00:39:33.841] Device 1 Address: 0000000000000000
> [2023-09-10 00:39:33.845] Device 2 Address: 0000000000000000
> [2023-09-10 00:39:33.849] Device 0 Resolution: 0
> [2023-09-10 00:39:33.849] Device 1 Resolution: 0
> [2023-09-10 00:39:33.853] Device 2 Resolution: 0
> [2023-09-10 00:39:33.853] Config of temperature sensors done
> [2023-09-10 00:39:33.857] CogTask init
> [2023-09-10 00:39:33.857] TempRefreshTask init
> [2023-09-10 00:39:33.861] starting
> [2023-09-10 00:39:34.472] The MAC address is: BBB
> [2023-09-10 00:39:35.472] AAA
> [2023-09-10 00:39:35.472] QQQQ
> [2023-09-10 00:39:35.472] ZZZZZZ
> [2023-09-10 00:39:35.472] MMMMMMM

End of report

RobertLRead, 2023-09-10:
It looks like in doing these runs you did not undefine the
 ETHERNET_BOARD_PRESENT
#define on line 71 in main.cpp

Please try to do it again, defining that constant to 0 (with the board
absent.)

The fact that it hangs with the board absent at start up is a known issue
that @Geoff Mulligan ***@***.***> is working on.

RobertLRead, 2023-09-10:
Data point: I've been running an empty Due with no Ethernet shield for about 18 hours with no hang.

ForrestErickson, 2023-09-10:
Modifying code to remove ethernet

> //#define ETHERNET_BOARD_PRESENT 1
>  #define ETHERNET_BOARD_PRESENT 0 //No ethernet.
> 

Resume testing about 20230910 1010

ForrestErickson, 2023-09-11:
Due running since 10:36 at 23:46.

ForrestErickson, 2023-09-12:
Lee stopped the run at 20230912 at about 14:00 and it was still running.

ForrestErickson, 2023-09-14:
Same as #223 
20230914_1752 Resume testing on MOM with [c79bc3c](https://github.com/PubInv/NASA-COG/commit/c79bc3c42ce0ccab385441132d64aecffc58d61a)

Still running as of 19:04 or over an hour.

ForrestErickson, 2023-09-18:
@gmulligan 
@RobertLRead 

### Summary of morning of 20230918 
Closing issue as the failure with Ethernet has been removed (cannot be reproduced by testing on MOM with [c79bc3c](https://github.com/PubInv/NASA-COG/commit/c79bc3c42ce0ccab385441132d64aecffc58d61a)).

Notes:
The Due with ethernet shield ran with out "halt"  from the afternoon of 14 September to morning of 18 September 2023.

> [2023-09-14 17:52:14.370] Starting Ox...
> [2023-09-14 17:52:14.473] Create watchdog (todo)
> [2023-09-14 17:52:14.473] Core booted...
> [2023-09-14 17:52:14.570] BEGINNING ASSERTION CHECKS!!
> [2023-09-14 17:52:14.570] IF YOU DO NOT SEE THE WORDS 'ALL CLEAR' BELOW AN ASSERTION HAS FAILED
> [2023-09-14 17:52:14.673] ALL CLEAR!!

to

> [2023-09-18 09:22:51.658] 
> [2023-09-18 09:22:51.658] Machine State: Off
> [2023-09-18 09:22:51.662] Target      C: 30.00
> [2023-09-18 09:22:51.662] Post Heater C: 22.75
> [2023-09-18 09:22:51.662] Post Getter C: 0.00
> [2023-09-18 09:22:51.666] Post Stack  C: 23.25
> [2023-09-18 09:22:51.666] Stack volts V: 0.00
> [2023-09-18 09:22:51.670] Stack amps  A: 0.00
> [2023-09-18 09:22:51.670] Stack ohms  O:  N/A
> [2023-09-18 09:22:51.674] Heater DC    : 0.0000
> [2023-09-18 09:22:51.674] Fan PWM      : 0.00
> [2023-09-18 09:22:51.674] Fan RPM      : 0.00
> [2023-09-18 09:22:54.169] Running ReadTemps
> [2023-09-18 09:22:54.169] About to _readTemperatureSensorsQQQ
> [2023-09-18 09:22:54.349] BBB
> [2023-09-18 09:22:54.427] Temp : Post Heater: 22.75
> [2023-09-18 09:22:54.427] All Temps, going backward in ms:
> [2023-09-18 09:22:54.431] 0 : 22.75
> [2023-09-18 09:22:54.431] 1000 : 22.75
> [2023-09-18 09:22:54.431] 2000 : 22.75
> [2023-09-18 09:22:54.435] 3000 : 22.75
> [2023-09-18 09:22:54.435] 4000 : 22.75
> [2023-09-18 09:22:54.435] 5000 : 22.75
> [2023-09-18 09:22:54.439] 6000 : 22.75
> [2023-09-18 09:22:54.439] 7000 : 22.75
> [2023-09-18 09:22:54.439] 8000 : 22.75
> [2023-09-18 09:22:54.439] 9000 : 22.75
> [2023-09-18 09:22:54.445] 10000 : 22.75
> [2023-09-18 09:22:54.445] 11000 : 22.75
> [2023-09-18 09:22:54.445] 12000 : 22.75
> [2023-09-18 09:22:54.447] 13000 : 22.75
> [2023-09-18 09:22:54.447] 14000 : 22.75
> [2023-09-18 09:22:54.447] 15000 : 22.75
> [2023-09-18 09:22:54.447] Done with _readTemperatureSensorsError: Could not read temperature data: 2
> [2023-09-18 09:22:54.529] Bad post_getter_C
> [2023-09-18 09:22:54.566] 
> [2023-09-18 09:22:54.566] Machine State: Off
> [2023-09-18 09:22:54.570] Target      C: 30.00
> [2023-09-18 09:22:54.570] Post Heater C: 22.75
> [2023-09-18 09:22:54.575] Post Getter C: 0.00
> [2023-09-18 09:22:54.575] Post Stack  C: 23.25
> [2023-09-18 09:22:54.575] Stack volts V: 0.00
> [2023-09-18 09:22:54.579] Stack amps  A: 0.00
> [2023-09-18 09:22:54.579] Stack ohms  O:  N/A
> [2023-09-18 09:22:54.583] Heater DC    : 0.0000
> [2023-09-18 09:22:54.583] Fan PWM      : 0.00
> [2023-09-18 09:22:54.587] Fan RPM      : 0.00
> [2023-09-18 09:22:54.587] Target Temp : 30.00
> [2023-09-18 09:22:54.587] Good Temp Reads:116582
> [2023-09-18 09:22:54.591] Bad  Temp Reads:58291

closed 2023-09-18 by ForrestErickson