PubInv/NASA-COG #227: DB9 connector, wrong part number.

LokiMetaSmith, 2023-09-11:
[image]

Holes on front panel don't match with the DB9 connector

ForrestErickson, 2023-09-12:
Dang, I had thought I had checked exactly that. That the HEX nuts would mount them both.     

The part number should ahve had a "-4" rather than the suffix it did.

[image]

<html>
<body>

Digi-Key Part Number | A32117-ND |  
-- | -- | --
Manufacturer | TE Connectivity AMP Connectors |  
Manufacturer Product Number | 5747844-4 |  
Description | CONN D-SUB RCPT 9POS R/A SLDR |  
Manufacturer Standard Lead Time | 4 Weeks

</body>
</html>Digi-Key Part Number
A32117-ND
Manufacturer
[TE Connectivity AMP Connectors](https://www.digikey.com/en/supplier-centers/te-connectivity-amp)
Manufacturer Product Number
5747844-4
Description
CONN D-SUB RCPT 9POS R/A SLDR
Manufacturer Standard Lead Time
4 Weeks

ForrestErickson, 2023-09-12:
Mouser part number:

https://www.mouser.com/ProductDetail/TE-Connectivity/5747844-4?qs=x6EjVpvqMVOIOT3ErJFDOQ%3D%3D

5747844-4
TE Connectivity
Images are for reference only
See Product Specifications
Mouser #:571-5747844-4
Mfr. #:
5747844-4
Mfr.:[TE Connectivity](https://www.mouser.com/manufacturer/te-connectivity/)
Customer #:
Customer #
Description:
D-Sub Standard Connectors RECP FRNT MTL SHL 9P