PubInv/NASA-COG #234: Thermocouples still fail to read intermittently; seems related to fan

RobertLRead, 2023-09-12:
@ForrestErickson I would like to talk to you after you have read this.
I spent all today and last night working on the intermittent temperature read failures. I have tried to address it physically with various placements.

@LokiMetaSmith advanced the hypothesis that it was a timing problem. This may be true based on the use of PWM, but is not based on the blower tachometer interrupt as I suspected. I removed the interrupt comletely, and it make no discernible difference. I also performed a loop of 1000 math operations in the interrupt, and it made not discernible difference.

I have been peforming experiments, and recording  ( # bad readings / # good readings).

I have not taken an oscilloscope image since repairing and rebuilding the board; possibly that is the next step.

Here are some of my notes on experiments:
1. Without ever turning on the fan: 2/ 186
2. Long data in case: 52/179
3. Long data wire coiled: 14/115
4. Short data wire: 23/100
5. With no interrupt in place : 22/62
6. Changing PWM write to digital write of HIGH: 15/63
7. 100% fan speed: 20/100

I am at a loss as to the next step, and need to work on the networking code that Geoff gave me (and repair a wire in the case), so i am stopping this for now.

LokiMetaSmith, 2023-09-13:
So, the due has two dedicated PWM channels DAC0 and DAC1 might be worth
trying those pins,

In addition to PWM capabilities on the pins noted above, the Due has true
analog output when using analogWrite() on pins DAC0 and DAC1.

ForrestErickson, 2023-09-13:
Regarding, 

> "the Due has true analog output when using analogWrite() on pins DAC0 and DAC1."

To what purpose at this time?  Just FYI?

Further note. From https://store-usa.arduino.cc/products/arduino-due?selectedStore=us

> DAC1 and DAC2  These pins provides true analog outputs with 12-bits resolution (4096 levels) with the[analogWrite()](https://www.arduino.cc/en/Reference/AnalogWrite) function. These pins can be used to create an audio output using the [Audio library.](https://www.arduino.cc/en/Reference/Audio)
> Please note that DAC output range is actually from 0.55 V to 2.75 V only.

LokiMetaSmith, 2023-09-13:
If the issue is replated to the fan, we really need to know more.

   - is it the hardware fan, such that removing  the PWM cable removes the
   problem
   - is it  a quirk with how the arduino abstracted out the PWM timer
   configuration when it forces everything to use the analogWrite() for
   everything
   - DAC0 and DAC1 use different hardware inside the atmel SAM processor
   for generating clocks, so maybe it's a collision or reconfiguration of a
   clock by another library

RobertLRead, 2023-09-14:
If the 24V power to the fan is completely discoconnected, we get 2/286 bad reads.

This seems to place the blame squarely on the power and/or the switching thereof.  One would have thought the 110V AC would be more likely to do this, but the evidence says not.

I am unsure of the next step.

ForrestErickson, 2023-09-14:
I am accumulating the results of further experiments in issue #217

LokiMetaSmith, 2023-09-14:
we may need a separate power supply just for feeding the onewire circuit.
this would let us isolate noise from the DC bus, but as Lee would probably
point out, until we track down the source of the problem, we can't really
propose a solution.

ForrestErickson, 2023-09-14:
Regarding, "we may need a separate power supply just for feeding the onewire circuit."
FYI I have connected the Vin pin to 3.3V.  Perhaps it should be connected to the +5.  There is a single 4K7 resistor pulling up to 3.3V.  Again, I am accumulating the results of further experiments in issue https://github.com/PubInv/NASA-COG/issues/217 #217.

ForrestErickson, 2023-09-15:
Examining read_temps_task.cpp
In the function _readTemperatureSensors()
The counter " bad_temp_reads++;" for temperature errors is common to all three sensors.
If the errors are concentrated on one sensor we are blind to that with this reporting.

ForrestErickson, 2023-09-15:
Refereeing to notes in #217 there was no correlation between fan on or off with the errors on the Mockup Of Maryville unit.

ForrestErickson, 2023-10-04:
@RobertLRead 
What is the status of this as a function of the library version?
Can we close this?

CC: @LokiMetaSmith

closed 2023-10-09 by RobertLRead