PubInv/NASA-COG #235: Incident: Robs Due quits reading temperature, and is hot to the touch...

RobertLRead, 2023-09-13:
At about 7:00 pm on Tuesday night, I was working as normal.  I had just made a code commit. I had gotten off the phone with Lee. The thermocouple read quit working.  The bottom of the Due (which is exposed in the carrier) was hot to the touch. All other functions may have been working.

I am trying to remember if I did anything either to the software or hardware that changed things in anyway. I don't think I did, unless it was to unscrew the ground wire of the fan.

When I plug the thermocouple board into a spare Due, the temperature works fine.

RobertLRead, 2023-09-13:
The current system is running the fan all the time. I don't know why. The behavior when in off mode is not to run the fan. The code does not think it is sending a PWM signal.

RobertLRead, 2023-09-13:
The Due was holding 4V on pin 9, the FAN PWM pin. This should not even be possible. I suspect this means the Due is damaged. I have a spare that I can use to replace it in the morning.

This is very bad because I have no explanation for what caused this, unless it was unplugging the 24V negative wire from the fan controller. This means our system may destroy DUEs at random.

ForrestErickson, 2023-09-13:
Regarding, "unless it was to unscrew the ground wire of the fan."
If the fan ground wire is disconnected, positive voltage into the fan will still try to flow through the PWM pin.  The equivilant circuit is something like +24V / the eqivilant of the resistors internal to the fan and on the PWM pin. 

While I can only guess at what the Due technology does, in years past over voltaging an IO pin triggers the ESD protection diodes to become a parasitic SCR (Silicon controlled rectifier) which shunts the Vcc to GND internally through the ESD protection structure for that pin.  The only current limit would be possibly the 500 mA polyfuse on the Due. 

On the Controller v1 we have a buffer circuit which would prevent this current path.  It is the Q1 circuit in this screen capture.
**NOTE THAT THIS CIRCUIT INVERTS THE PWM and so with the Controller v1 the firmware must take this inversion into account.**

[image]

### Bottom line. Do not disconnect the negative line on the fan.   
To disable the fan, disconnect the +24 line.

RobertLRead, 2023-09-13:
This is good to know; possibly it explains my problem.

RobertLRead, 2023-09-13:
I'm trying to understand how 24V would ever go into a GPIO pin in this
situation.

I guess the transistor used to drain power inside the fan doesn't really
work exactly like a switch, and that if the drain is not connected, the
voltage can somehow rise on the gating function?

LokiMetaSmith, 2023-09-13:
it depends on how you have it wired.

if the 24v and 12v have a common ground, then unhooking the ground
reference from the fan, doesn't remove the ground reference at the control
board.

internally the fan likely has the equivalent of a voltage divider, so when
you remove the drain/ground pin in a voltage divider, the full voltage
potential is seen on the normally fraction of the full voltage, center pin

if the grounds are not tied together, then the potential is floating, and
there would be no damage, unless the 24v breaks down an insulation barrier,
which is unlikely

We should look into galvanically isolating our real world inputs/outputs
from the micrprocessor

RobertLRead, 2023-09-14:
The 12V and 24V grounds are tied together. You told me to do that; I think it is necessary because you want the ground for the Arduino to be the same as the ground for the fan.

Under the assumption that this was caused be me removing the GND wire instead of the 24V wire while the power was on, I'm closing this issue.

closed 2023-09-14 by RobertLRead

ForrestErickson, 2023-09-14:
I too think this should be closed.