PubInv/NASA-COG #240: refactoring ToDo checklist

LokiMetaSmith, 2023-09-13:
- [ ] expand MAX31855 SensorConfig class such that the SensorConfig abstract class allows for definition through SensorConfig &config the pin definitions of the SPI bus used by the MAX31855 thermocouples, as opposed to how they are now hard defined with #define statements. 
- [ ] need to refactor machine_core_def.h and move ODECS defines into a HAL similar to stage2_HAL
- [ ] refactor platformio.ini such that we can fully isolate libraries and defines when not in use. for example, onewire isn't used in stage2_heater, but it has to be included because of all the unused headers and c file defines
- [ ] Add in MAX31855 error checking into the readTemperature function to verify that the thermocouple amp is still operating as expected.

RobertLRead, 2023-10-09:
@LokiMetaSmith I think this is irrelevant now, correct?

LokiMetaSmith, 2023-10-09:
Correct

closed 2023-10-09 by LokiMetaSmith