PubInv/NASA-COG #252: Turning on a separate 24V fans drops the temperatures by half a degree C....

RobertLRead, 2023-09-15:
This is has got to be important; though how it relates to our missing values I don't know.

[image]

ForrestErickson, 2023-09-15:
Have I correctly annotated the graphic?
[image]

How did you switch the fan on and off?
To where did you connect the fans?  A photo would help.

On the MAX31850 amplifier boards there is a pin labeled 3V3.
During the temperature measurement I have measured the voltage there to drop a bit which I am guessing is an indication of the internal operation duration.  A similar measurement on your system might give insight for when the second fan is on and off.

Here is oscilloscope of some waveforms of the Mockup Of Maryville.
> [image]
> **CH1 is data, CH2 is the raw 3V3 of the Due, CH3 is TC03V3 and CH4 is TC13V3**
>

RobertLRead, 2023-09-15:
I attached the second fan with alligator clips.

RobertLRead, 2023-10-09:
@ForrestErickson I am not sure this is worth fixing, but your recent experience scoping these thermocouples may give us new insight. I think we should close this as "wontfix" but I want your opinion first.