PubInv/NASA-COG #255: Thermocouple reads fail when code compiled with platformIO but not Arduino IDE

RobertLRead, 2023-09-15:
This only occurs frequently when the fan is on. It is thus a combination of software and hardware.

The code:

https://github.com/PubInv/NASA-COG/tree/develop/experimentation/Dallas_Tester_Rob

Has (almost) the same code for platformIO (the src/main.cpp). With the fan on, it consistently (though intermittently) gives bad thermocouple reads. When the same code is compiled with the Arduino IDE, it has never been observed to fail.

This is a very important learning. @ForrestErickson @LokiMetaSmith @gmulligan

RobertLRead, 2023-09-15:
This has been determined to be because the Arduino IDE and PlatfromIO pull in different library version numbers which are the root cause of this problem.

A branch has been created for additional investigation:

investigateDallasTemperatureLibIssues

I am for the time being assigning this to @ForrestErickson to finish up and write the bug report, but he may of course assign this back to me.  Additional comments below will describe the problem in greater detail.

RobertLRead, 2023-09-15:
Dear @ForrestErickson,

Please observe two directories:

[https://github.com/PubInv/NASA-COG/tree/investigateDallasTemperatureLibIssues/experimentation/Dallas_Tester_Rob](https://github.com/PubInv/NASA-COG/tree/investigateDallasTemperatureLibIssues/experimentation/Dallas_Tester_Rob)

and 

[https://github.com/PubInv/NASA-COG/tree/investigateDallasTemperatureLibIssues/experimentation/Dallas_Tester_Via_IDE/Dallas_Tester_Rob](https://github.com/PubInv/NASA-COG/tree/investigateDallasTemperatureLibIssues/experimentation/Dallas_Tester_Via_IDE/Dallas_Tester_Rob)

The first is a typical "platformio" directory tree. In it there is "makefile" with a single target, "run". This will make and run the code in main/src.cpp.

This is exactly the same code (at present) in the .ino file in the other directory, which is in the typcial Arduino IDE style. 

This small program (based on your work) does an anlog write on pin 9 to set up a PWM signal for our SanyoACE fans, which on my hardware manifests bad thermocouple reads up to 30% of the time---when the DallasTemperature version is 3.11.0.

In the platform.ini file in the first directory, you will find where I commented out this problem.

In the Arduino IDE, which library is installed is local---you can change it wil the tool manager. You may or may not be able to get it to install 3.11.0 --- herein lies the rub. Platform.io seems to be willing to get the lastest verison in GitHub, the Arduino IDE wants to get the latest RELEASE---probably the better behavior.

In any case, the platform.io version can be used to perform a red/green test of the problem (at least, for me, because I have an OEDCS which manifest the problem.) It may work for you, because you may never have installed the wrong library yet!

So, this small program clearly manifest the problem (though my coding of it is very messy, I was hacking and slashing.)

I would like you to try to duplicate this and then submit a bug report to DallasTemperature.

ForrestErickson, 2023-09-15:
Lee has reported thermocouple failures on the Mockup Of Maryville hardware WITH OUT THE FAN running.
This is a report for the system running for about a day.
[image]

Note that until about an hour ago oscilloscope probes were hanging on the + and - terminal of the TC0 amplifier.

RobertLRead, 2023-09-15:
Does that mean it ran for 24 hours without a hang?

ForrestErickson, 2023-09-21:
20230921 Some more insights.  The  15 Sep report where the MOM had errors is probably because the firmware was expecting three TC and only finding two and so counting an error for ever cycle of read.

The ratio of 28656/14328 is exactly 2.

ForrestErickson, 2023-09-21:
Lee examined the  milesburton Arduino-Temperature-Control-Library and sees that the keywords file at
https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/keywords.txt
has key words for all of the MAX31850 faults.

DEVICE_DISCONNECTED_C	LITERAL1
DEVICE_DISCONNECTED_F	LITERAL1
DEVICE_DISCONNECTED_RAW	LITERAL1
DEVICE_FAULT_OPEN_C	LITERAL1
DEVICE_FAULT_OPEN_F	LITERAL1
DEVICE_FAULT_OPEN_RAW	LITERAL1
DEVICE_FAULT_SHORTGND_C	LITERAL1
DEVICE_FAULT_SHORTGND_F	LITERAL1
DEVICE_FAULT_SHORTGND_RAW	LITERAL1
DEVICE_FAULT_SHORTVDD_C	LITERAL1
DEVICE_FAULT_SHORTVDD_F	LITERAL1
DEVICE_FAULT_SHORTVDD_RAW	LITERAL1

However examining the code at the DallasTemperature.cpp files shows no tests for 

For DEVICE_DISCONNECTED there are 17 hits.
For DEVICE_FAULT_OPEN there are no hits
For DEVICE_FAULT_SHORTGND there are no hits
For DEVICE_FAULT_SHORTVDD there are no hits.

The DallasTemp library is not supporting OPEN, SHORTGND, or SHORTVDD

ForrestErickson, 2023-09-21:
Search the Arduino Library manager for MAX31850
[image]
There are three hits.

The first is: https://github.com/RobTillaart/MAX31850

The second is 
Regarding, the library: https://github.com/adafruit/MAX31850_DallasTemp 
This library defines only one error code as follows
// Error Codes
#define DEVICE_DISCONNECTED -127 //!< Device disconnected error code

The .cpp file has seven location where DEVICE_DISCONNECTED is used and no other error codes are tested.

The third is the one we are using.

LokiMetaSmith, 2023-09-21:
on ODECS#2 we got the fan working last night and I didn't notice any issue
with dropping TC readings.

RobertLRead, 2023-09-21:
That is consistent with bad connections.

ForrestErickson, 2023-09-21:
I have loaded the library [RobTillaart](https://github.com/RobTillaart)
/
[MAX31850](https://github.com/RobTillaart/MAX31850)

This library has a function: 
uint8_t MAX31850::getErrorCode()
{
  return _errorBits;
}

I added this function to the example code: MAX31850_test_disconnect.ino

  sensor.read();
  float temperature = sensor.getTempTC();
  uint8_t  myErrorCode = sensor.getErrorCode();
  Serial.print("myErrorCode= ");
  Serial.println(myErrorCode);
  

I then experimented with disconnecting the TC, Shorting the TC to GND and to VCC and got unique error codes.

[image]

ForrestErickson, 2023-09-21:
I have elaborated the code.
```
sensor.read();
  float temperature = sensor.getTempTC();
  uint8_t  myErrorCode = sensor.getErrorCode();
  Serial.print("myErrorCode= ");
  Serial.print(myErrorCode);
  Serial.print(" ");

  switch (myErrorCode) {
    case 0:
      // statements
      Serial.println("Valid Temp");
      break;
    case 1:
      // statements
      Serial.println("Open Sensor");
      break;
    case 2:
      // statements
      Serial.println("Short GND");
      break;
    case 4:
      // statements
      Serial.println("Short VCC");
      break;
    default:
      // statements
      Serial.println("TC ERROR NOT KNOWN");
      break;
  }
```

[image]

LokiMetaSmith, 2023-09-22:
this has been resolved as a mechanical issue in @RobertLRead 's thermocouple connection

closed 2023-09-22 by LokiMetaSmith