PubInv/NASA-COG #259: COPI improperly attached to DO

LokiMetaSmith, 2023-09-16:
[image]
 COPI is attached to DO on the MAX31855 devices through R6
COPI is a Data OUTPUT pin, where DO is also a Data OUTPUT pin. 

DO should be hooked to CIPO

ForrestErickson, 2023-09-16:
## Rework for Controller v1 Thermocouple CIPO

Remove R6.
Solder jumper wire from test point CIPO1 to the pad of R6 nearest the thermocouples.  

[image]
Jumper wire indicated by the ruller in this screen shot.

LokiMetaSmith, 2023-09-22:
Because we're using 1-wire so far, I've forgotten that this is still relevant. 

This should be fixed for Control V2 and modified for the existing boards

ForrestErickson, 2023-09-22:
@LokiMetaSmith as you rework boards make notes of it in this issue or rework them all at once so that this issue can reflect and be one stop reference to the configuration(s).

LokiMetaSmith, 2023-10-06:
leaving this for control v2