PubInv/NASA-COG #261: control board V1 kills ethernet boards

LokiMetaSmith, 2023-09-16:
still diagnosing but I received two ethernet boards from Geoff and a new dueI verified that both worked and it turned out that the ethernet shield connected to the due was bad, it was an arduino brand board so I didn't expect that failure but failures happen. 

I also verified the ethernet board (seeed studio w5500) for the stage 2 heater and it still works, so before 10pm I had three good ethernet boards and one bad one. 

I then loaded up the ethernet address code and got the mac address from ODECS #2 using one of the Seeed studio w5100 that I received from Geoff, I then loaded up the latest firmware and noticed that while everything runs, the ethernet still wasn't coming up. Also the voltage on the USB line was being dragged down around ~3.7-3.8v. I plugged in a usb power supply that reported current and it read around 300mA, which is MUCH higher than expected. 

I have verified the steady state voltages on all pins and I can't explain why the ethernet boards are failing on the control board V1. 

The ethernet boards have voltage level shifters on them which "should" protect them from over voltage errors, and on inspection, the 3.3v regulator seems intact.

RobertLRead, 2023-09-16:
> Assigned #261 <https://github.com/PubInv/NASA-COG/issues/261> to
> @RobertLRead <https://github.com/RobertLRead>.

I assume this is an electrical problem --- I'm not sure what I can say
about it. I agree if the USB voltage is being pulled low something is
seriously wrong.

I suppose it must be possible to figure out what is drawing the current,
though it might require isolating devices in a way that might be difficult.

Are you looking for anything else from me?

LokiMetaSmith, 2023-09-16:
curiouser and curiouser

When plugging in USB the first time, I can not get reliable ethernet cable detection even after giving it several seconds to settle. 

After a single Reset event, the ethernet cable connection works as expected. 

This may be giving me false failure conditions, although I would say that this would count as an actual failure condition.

LokiMetaSmith, 2023-09-16:
Ok, at the end of the night, I have two dead arduino ethernet boards, the voltages look ok, but I suspect the logic bus interface  chips are damaged. I have two good ethernet boards, one bad DUE which seems locked in perpetual reset, and two good (for now) Due, to continue testing with.

LokiMetaSmith, 2023-09-16:
I've verified the circuit on the SPI bus and the part selection looks good, no wrong parts, etc... The unfortunate thing is that unless the problem can be identified, ODECS no.2 can't be used for evaluation until the problem is discovered

RobertLRead, 2023-09-16:
The OEDCS #2 can alway be wired by hand after our return from Utah. It is a
nice-to-have for this purpose.  You've already done a lot of the major
wiring of the heavy components, it is only the GPIO-pin wiring that would
need to be redone and that can be done in a couple of hours.

ForrestErickson, 2023-09-16:
@LokiMetaSmith regarding the text in at the beginning of the issue. When you can, I would like to review and edit for clarity of the sequence of events to remove any ambiguity. 
Lee

ForrestErickson, 2023-09-16:
## Electrical Interface, Due to Seeds Ethernet Shield V2.0
The connection from the Due into the Ethernet Shield V2.0, W520 shield is through the SPI interface which is a straight through connection through the Controller v1 assembly.

Although we make connections to the SPI on thermocouple amplifiers, our **Controller v1** makes no changes to the connection to the Ethernet shield merely passing it through.  

I believe this is the documentation for the Seeds boards we have: https://wiki.seeedstudio.com/Ethernet_Shield_V2.0/

### UNO Form Factor, 5V?
The form factor is of the UNO which is a 5V device. The documentation makes no mention of the interface voltage levels and I infer 5V interface is assumed. 

### Measurement on MOM of CIPO
[image]

The CIPO signal from the W520 into the Due shows the signal ever so slightly above the 3V2 supply.  Possibly 200 mV. 
**This indicates the signal is pulled above the Vcc of the Due which is dangerous.**  

I am left with the question, "Are there ethernet shields in the Due form factor designed with 3V3 interface?"

ForrestErickson, 2023-09-16:
Found schematic for the Ethernet board.  
https://files.seeedstudio.com/wiki/Ethernet_Shield_V2.0/res/Ethernet%20Shield%20v2.0%20sch.pdf

### 3V3 Interface
It appears as if only 3V3 is going into the W520 IC so probably OK to use this board with Due. Ie, this board will not over voltage the Due input.

LokiMetaSmith, 2023-09-16:
Due GIPO pins are rated for  1.62V to 3.6V, per page 8 of
https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf

LokiMetaSmith, 2023-09-16:
One concern is that the level shifter circuit puts a pull up on all SPI
pins, It's suspected that somehow because the level shifter internally is
parallel with the isolation circuit and the Due hardware SPI lines, that
the SN74lv buffer IC has been damaged. However, this would be difficult to
probe direclty

[image: image.png][image: image.png]

ForrestErickson, 2023-09-16:
Regarding, " I have two dead arduino ethernet boards,"  
Exact make and model number please.  
And as we can anticipate, exact make and model number of the not (yet?) dead Ethernet assemblies.

In the MOM I have (IIRC) Seeds Studio V2.0. Right now it will be very difficult to unmount the board as it is under the prototype shield with wires restricting the movement.

ForrestErickson, 2023-09-16:
Regarding, "SN74lv buffer IC"  From where did you get this IC type?  I looked but did not see on the Ethernet schematic. It would not be the first time I could not see the obvious.   And "SN74lv buffer IC" is not a full part number which is about what I would like to learn more.

LokiMetaSmith, 2023-09-16:
https://www.arduino.cc/en/uploads/Main/arduino-Ethernet-Shield2-V2-sch.pdf

https://files.seeedstudio.com/wiki/W5500_Ethernet_Shield_v1.0/res/W5500_Ethernet_Shield_v1.0.pdf

https://files.seeedstudio.com/wiki/Ethernet_Shield_V2.0/res/Ethernet%20Shield%20v2.0%20sch.pdf

These are the three versions of "arduino ethernet shield" we have floating
around.

each is slightly different, all of them have some sort of buffer, none of
them guarantee a part number and may differ based on lot.

ForrestErickson, 2023-09-16:
Lee has the Seeeds V2 and that is what was (apparently) offered on Amazon.   Suspicious minds will always check.

ForrestErickson, 2023-09-16:
## Circuit Loading SPI Due and Ethernet

Just pondering, the level shifter circuit in the schematic below is connected to and loading the SPI interface.
[image]
This is the same interface circuit used on the ESP32 PolyVent level shifter which stems from the classic I2C level shifter between 3V3 and 5V logic.

**Bottom line. I do not see anything alarming or likely to cause malfunction.**

ForrestErickson, 2023-09-16:
@LokiMetaSmith 
Regarding, " However, this would be difficult to probe direclty" above.

Here is how to probe. An "Octopus Curve Tracer" will let you test for common failure modes and symptoms.
See: https://www.youtube.com/watch?v=Gwo3pEH7hUE

If you have an oscilloscope and a function generator you can make up one.  Connecting the DUT so that the likely common earth ground of function generators and oscilloscopes is a mental challenging exercise you can avoid if you have the isolation transformer which is used in this video.

If you have and are willing to cut up an older transformer based wall supply to get to the AC output you will have such an isolation. The exact value of the output voltage is not critical as you will put a resistor in series which you can choose.

For an IC you will short the Vcc pins to GND pins and probe looking for the typical ESD protection diodes.

My OCTOUNO project has images of what the ESD diodes curve trace should look like. See:
https://github.com/ForrestErickson/OctoUNO

LokiMetaSmith, 2023-09-19:
one of the "damaged" ethernet shields, Arduino Ethernet shield works with a arduino UNO, which is a 5v part. I suspect that the DUE is not five volt tolerant and maybe that's the issue

LokiMetaSmith, 2023-09-19:
[image]

there's no filtering or level shifting on the SPI lines on the DUE

LokiMetaSmith, 2023-09-19:
tried the remaining working ethernet board with a working due, the pair works, but the moment the spare control board v1 is inserted between them, the ethernet board doesn't work. I've confirmed that the ethernet survived a brief power on cycle and test with the control board. 

next step is to remove all unrelated circuitry on the SPI bus.

ForrestErickson, 2023-09-19:
@LokiMetaSmith 
Regarding, "next step is to remove all unrelated circuitry on the SPI bus."
Confirm please that you mean, "next step is to remove all unrelated circuitry on Controller v1 board connected to the SPI bus."

### Regarding that circuitry.
For CIPO that is only R62 a 1K pull up to 3V3 and Q6
For SCK that is only R64 a 10K pull up to 3V3 and Q7 AND R5 to the thermocouples.
For COPI that is only R65 a 10K pull up to 3V3 and Q8 AND R6 to the thermocouples.
For COPI_nCS that is only R61 a 10K pull up to 3V3 and Q5

If you remove R5 and R6 you will need to devise another, independent,  SPI clock and COPI for the thermocouples.
The are located here.
[image]

LokiMetaSmith, 2023-09-19:
it was easier to solder on extra headers onto a blank PCB than it was to remove parts on an existing board. 

On a blank pcb with just the headers, I have been successful with a known good ethernet board and a known good DUE to get DHCP access and verify functionality of the control board.

LokiMetaSmith, 2023-09-19:
[image]

I have now successfully gotten a populated control v1 board with a DUE and ethernet shield to work.
Doing so required cutting all traces to SPI control level shifter

LokiMetaSmith, 2023-09-20:
proposed cut lines for I2C level shifter
[image]

LokiMetaSmith, 2023-09-20:
It was observed that removing the resistors seperating the 3.3v and 5v local from the bus voltage didn't allow functionality. except in the case where a UNO was used in place of a DUE, but the UNO is a 5v tolerant part on all GPIO lines

ForrestErickson, 2023-09-20:
When time allows, 
Please add a serial number to each so that they will never be mistaken for one another.
Please make a photograph of the several V1 assemblies so Lee can see all of the components populated on the assemblies.

ForrestErickson, 2023-09-25:
Trying to ponder the Ethernet Shield problem.

### For the MISO signal 
Where the Ethernet shield is the low impedance drive to the net,
I examined the schematic for the Arduino Etherntet Sheild V2.  I see that the 3.3V MISO from the W5200 is fed to the header where it will connect to the UNO / DUE.  
https://www.arduino.cc/en/uploads/Main/arduino-Ethernet-Shield2-V2-sch.pdf

The Seeed studio V1 also has  3.3V MISO from the W5200 is fed to the header.
The Seeed studio V2 also has  3.3V MISO from the W5200 is fed to the header. 
The data sheet web page does not say 3.3V interface. https://www.seeedstudio.com/W5200-Ethernet-Shield-p-1577.html

The  W5500 Ethernet Shield v1.0 specification at: https://www.seeedstudio.com/W5500-Ethernet-Shield.html
Specifically says "3.3V operation with 5V I/O signal tolerance"

### For SCK, SS (nCS)  and MOSI signal  
The Due is expected to be the low impedance drive the only loads appear to be logic inputs and on the Controller V1 board the same level shifter circuit we put on the PolyVent GPAD interface card which worked with the ESP32.

### Circuit Simulations
Just incase I was not understanding the level shifter circuit function I wokled upa Falstad simulation
Here with two switches open to GND  and showing the logic levels on either side are at approprate levels.
[image]

Next close the 3.3V switch to simulate a Due driving the signal low.
[image]
This looks as expected.

Next close the 5V switch to simulate a 5V logic signal low.
[image]

Again, this looks as expected.

ForrestErickson, 2023-11-09:
20231109 Begin testing with Control V1.1 SN9 assembly. See also notes in https://github.com/PubInv/NASA-COG/issues/379
Copied from there.
Connect the DUE to the Ethernet WITH OUT the Control V1.1.
Update local repository clone and load firmware.
Set for "-D ALLOW_BAD_THERMOCOUPLES_FOR_TESTING". Compile and load.
It boots and reports version "Starting OEDCS; Rev: 0.3.10"
DUT got network time.

Connect Due to Control V1.1 assembly.
Apply power and connect with Terminal
Normal boot messages and the expected ethernet failure

12:35:30.744 -> W5x00 init failed
12:35:31.354 -> W5x00 init failed
12:35:32.010 -> W5x00 init failed
12:35:32.621 -> W5x00 init failed
12:35:33.234 -> W5x00 init failed
12:35:33.842 -> W5x00 init failed
12:35:34.450 -> W5x00 init failed
12:35:35.106 -> W5x00 init failed
12:35:35.714 -> W5x00 init failed
12:35:36.323 -> W5x00 init failed
12:35:36.323 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
12:35:41.344 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
12:35:46.350 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!

Connect the Ethernet Sheild
Apply power. Connect USB port. Open Terminal window
the DUT found the thermocuples and continued but did not find the Ethernet sheild
12:40:10.379 -> CogTask init
12:40:10.379 -> OEDSCSerialTask Inited
12:40:10.998 -> W5x00 init failed
12:40:11.604 -> W5x00 init failed
12:40:12.238 -> W5x00 init failed
12:40:12.828 -> W5x00 init failed
12:40:13.483 -> W5x00 init failed
12:40:14.080 -> W5x00 init failed
12:40:14.731 -> W5x00 init failed
12:40:15.333 -> W5x00 init failed
12:40:15.954 -> W5x00 init failed
12:40:16.590 -> W5x00 init failed
12:40:16.590 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
12:40:21.571 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!

ForrestErickson, 2023-11-09:
## Troubleshooting Ethernet
Instrument the interface to the Ethernet card with oscilloscope

CH1 is D10 the nCS
CH2 is SPI_SCK
CH3 is COPI1
CH4 is CIPO1
[image]

The CIPO1 signal is not going to full amplitude, Not going low.

Conjecture the Ethernet sheild is not able to pull the 1K resistor R62 low.
[image]

Remove R95, the 1R0 resistor to disconnect the R62 (and Q6)
[image]

Reboot the system.  The DUT found the ethernet and internet.

13:13:54.177 -> OEDSCSerialTask Inited
13:13:54.737 -> MAC Address: FE:ED:16:6B:72:67
13:13:56.377 -> Unix time = 1699553636
13:13:56.377 -> The UTC time is 18:13:56
13:13:56.377 -> Network started
13:13:56.377 -> 
13:13:56.377 -> 
13:13:56.897 -> Duty Cycle Setup
13:13:56.897 -> DutyCycleTask init

The CIPO line now goes low.
[image]

Conclusion. The CIPO line on the ethernet card is not able to sink enough current to make valid logic levels.

ForrestErickson, 2023-11-09:
## Short Term Countermeasure
Unsolder R95 on the V1.1 control assemblies. The SPI interface on the RJ45 will not be available for the GPAD.

## Long Term Fix. 
Implement and independent SPI on pins unrelated to the Ethernet shield.

LokiMetaSmith, 2023-11-10:
Does this mean the boards are ready to be shipped?

ForrestErickson, 2023-11-10:
Not yet ready.
I have been very slow on this because other activates (Some Public Invention and some personal) have absorbed my energetic hours during the past week.
Tomorrow is not looking good.