PubInv/NASA-COG #262: when DUE is held in reset, ALL GPIO pins are PULLED HIGH

LokiMetaSmith, 2023-09-16:
this is somewhat unexpected as most MCU's go high impedance when held in reset.

LokiMetaSmith, 2023-09-16:
[image]
per the  Atmel SAM datasheet, this is expected, PIO Input and Internal pull-up enabled. 

This means that when the RESET button is pushed, ALL heaters should turn on.

RobertLRead, 2023-09-16:
I do not observe the SSR coming on when I reset the Due on the OEDCS #1.

ForrestErickson, 2023-09-16:
## Safe in Reset by Design, Controller v1
The Data sheet does not show the value of the internal pullup bit from this Arduino web site is the following
https://docs.arduino.cc/learn/microcontrollers/digital-pins

> The value of this pullup depends on the microcontroller used. On most AVR-based boards, the value is guaranteed to be between 20kΩ and 50kΩ. On the Arduino Due, it is between 50kΩ and 150kΩ. For the exact value, consult the datasheet of the microcontroller on your board.

### Controller v1 PWM and SSR Reset
Anticipating that the processor could be in reset indefinitely there are resistors on the Controller v! PCB which set the desired behavior.

* For the SSR this should be OFF.
Here is the circuit. R81 makes the SSR OFF.
[image]

* For the blower we want OFF. The PWM at the blower should be low so Q1 must be ON which R41 makes happen if the Due is in reset.
[image]

RobertLRead, 2023-11-17:
@ForrestErickson can this issue now be closed?