PubInv/NASA-COG #263: Stage 2 code is broken after integration of flash memory code

LokiMetaSmith, 2023-09-16:
Stage2 wouldn't compile unless the libary 	sebnil/DueFlashStorage was defined, but while writing the program, a very strange wall of read/write operations makes me believe there's something very very very wrong with the code. 

writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xc9e00)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xc9f00)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xca000)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xca100)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xca200)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0xca300)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)

[==========================    ] 89% (1188/1326 pages)writeWord(addr=0x20001028,value=0x20002000)

It did this until it reached 100% at which point it entered in a normal run loop

RobertLRead, 2023-09-16:
I am not seeing this occur. What command are using to produce this?

LokiMetaSmith, 2023-09-16:
pio test -e due_stage2_heater -f "test_stage2_heater" -vvv 

This command worked as expected previously, I also only see that block of text when programming, not on first boot.

RobertLRead, 2023-09-16:
This Arduino seems relevant:

https://forum.arduino.cc/t/cant-upload-to-due-anymore-pls-help/534743

We have the same code at the beginning:

Planes       : 2
Lock Regions : 32
Locked       : readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
....

But in this issue they can't upload; we don't seem to be experiencing the same problem. In fact it remains unclear if this is a problem at all; these are not necessarily error messages.

However, I pulled code from 3 days ago and built in verbose mode and did not see this. So it may be that writing to flash is causing this, whether it is a problem or not.

RobertLRead, 2023-10-09:
This voluminous situation is not known to create a problem. I'm closing this as wontfix for now.

closed 2023-10-09 by RobertLRead