PubInv/NASA-COG #267: Caught the network_udp class in a hang...

RobertLRead, 2023-09-16:
I just had about 4 hangs in a row after about 5 minutes of running oedcs, and, turning on verbose logging, caught the OEDCSNetworkTask in a hang. The following debugging shows it (not all of these statments are in the development branch yet:)

dutyCycle
About to Run task!
dutyCycle
Finished Run!
getNextTask
-4546
-5988
-6116
-150
-113878
8
0
-53988
nexTask
OEDCSNetwork
About to Run task!
OEDCSNetwork
The NetworkUDPTask was run

Seeking Params

Here is the relevant code:

```
    if (DEBUG_UDP > 1) {
      Serial.println("Seeking Params");
      delay(50);
    }
    bool new_packet = NetworkTask::net_udp.getParams(3000);
    if (DEBUG_UDP > 1) {
      Serial.println("Done with Params!");
      delay(50);
    }
    if (new_packet) {
      if (DEBUG_UDP > 1) {
        Serial.println("Got a Param Packet!");
      delay(50);
      }
```

I conjecture that the only way this can happen is if parsePacket blocks, and there is awarning in the relevant code about this:

From here: [https://github.com/arduino-libraries/Ethernet/blob/39103da0e1bc569023625ee4693272773397dbb6/src/EthernetUdp.cpp](https://github.com/arduino-libraries/Ethernet/blob/39103da0e1bc569023625ee4693272773397dbb6/src/EthernetUdp.cpp
)

Note: I have not yet enabled watchdog resets, but we need to solve this problem in any case. We may have to hack their libraries. Since I can duplicate it now, i will continue debugging this.

```
int EthernetUDP::parsePacket()
{
	// discard any remaining bytes in the last packet
	while (_remaining) {
		// could this fail (loop endlessly) if _remaining > 0 and recv in read fails?
		// should only occur if recv fails after telling us the data is there, lets
		// hope the w5100 always behaves :)
		read((uint8_t *)NULL, _remaining);
	}

	if (Ethernet.socketRecvAvailable(sockindex) > 0) {
		//HACK - hand-parse the UDP packet using TCP recv method
		uint8_t tmpBuf[8];
		int ret=0;
		//read 8 header bytes and get IP and port from it
		ret = Ethernet.socketRecv(sockindex, tmpBuf, 8);
		if (ret > 0) {
			_remoteIP = tmpBuf;
			_remotePort = tmpBuf[4];
			_remotePort = (_remotePort << 8) + tmpBuf[5];
			_remaining = tmpBuf[6];
			_remaining = (_remaining << 8) + tmpBuf[7];

			// When we get here, any remaining bytes are the data
			ret = _remaining;
		}
		return ret;
	}
	// There aren't any packets available
	return 0;
}
```

RobertLRead, 2023-09-16:
Upon further debugging this is NOT haning in the network call but in a debugging statement that uses the "String" (Capital S) clas.

I am still working on it.

RobertLRead, 2023-09-17:
This is not a network problem as I suggest. Further debugging shows it is likely a buffer overrun or pointer dereference cause by parsing the machine script which we were downloading. Why it shows up now I don't know. I have not been able to identify, but I believe it goes away if don't request the params.

RobertLRead, 2023-09-18:
As of now I have removed the parsing from the code; long-term we need to put it back in. Since Geoff has changed the code some more to remove memory leaks, this is worth retesting.

LokiMetaSmith, 2023-10-06:
@RobertLRead when you get a chance, please review this to see if it can be closed

closed 2023-10-06 by RobertLRead