PubInv/NASA-COG #270: Finding the IP Address Once I Know the MAC Address on OEDCS

ForrestErickson, 2023-09-18:
I have closed the issue regarding how to find the MAC address on my Due Ethernet system MOM.
It is: Labeling, Identifying MAC on the OEDICOS #84

Now that I have the MAC address, how do I find the IP address on my network?

ForrestErickson, 2023-09-18:
My MAC address on the MOM is: FE:ED:16:6B:72:67

On the ESP32 projects with which I worked the reference project code reported the IP address out a serial port, for example.   I also wrote some code to display the IP address on an LCD. 

The current OEDCS code does not help us with an IP address.

So I went about it another way.

The "arp -a" command will show all devices on the network with IP address and MAC address. However you cannot  arp to a MAC address and it will only show if there has been local traffice.. 

The solution is to ping ever address on the LAN and then do the "arp -a"

The following DOS CMD window command will serial ping the full address space on my LAN
`for /L %i in (1,1,255) do @ping -n 1 -w 1 192.168.1.%i`

Once I did this 

> C:\Users\Admin>arp -a
> 
> Interface: 192.168.1.118 --- 0x11
>   Internet Address      Physical Address      Type
>   192.168.1.1           e0-46-9a-26-f0-1b     dynamic
>   192.168.1.50          00-60-b0-44-be-5a     dynamic
>   192.168.1.108         90-02-a9-3b-86-a1     dynamic
>   192.168.1.130         a0-d2-b1-29-25-cd     dynamic
>   192.168.1.133         b8-27-eb-31-25-57     dynamic
>   192.168.1.136         fe-ed-16-6b-72-67     dynamic
>   192.168.1.252         30-23-03-f0-32-21     dynamic
>   192.168.1.254         64-52-99-52-26-f7     dynamic
>   192.168.1.255         ff-ff-ff-ff-ff-ff     static
>   224.0.0.22            01-00-5e-00-00-16     static
>   224.0.0.251           01-00-5e-00-00-fb     static
>   224.0.0.252           01-00-5e-00-00-fc     static
>   239.255.255.250       01-00-5e-7f-ff-fa     static
>   255.255.255.255       ff-ff-ff-ff-ff-ff     static
> 

So the IP address of the MOM is: 192.168.1.136

LokiMetaSmith, 2023-09-18:
we should report in the console the assigned IP address on boot

ForrestErickson, 2023-09-18:
I knew from a previous coworker that you can get a faster ping of all addresses if you fire off code which opens a new thread for each PING. 

I asked ChatGPT for such code and this was suggested as a batch file

```
@echo off
for /L %%i in (1,1,255) do (
    start cmd /c ping -n 1 -w 1 192.168.1.%%i
)
```
[PING_ALL.zip](https://github.com/PubInv/NASA-COG/files/12652330/PING_ALL.zip)

I made such a file and ran it.  
The result is for some moments alarming as 255 CMD windows open up
[image]
 but they all close with in the time out and the wait was much shorter than the for do loop above. 

I experimented with pinging the MOM unit with and with out and then with Ethernet cable connected and MOM successfully reconnected and responded to ping.

[image]

ForrestErickson, 2023-09-18:
Closing the issue since I have a solution.
I have labeled the MAC and IP address on the MOM hardware.
[image]

closed 2023-09-18 by ForrestErickson