PubInv/NASA-COG #273: Report Independent Thermocouple Failure Counts,

ForrestErickson, 2023-09-18:
## Report independently for the three thermocouples the counts. 
Report on one line the count for good
Report on a second line the count for bad.

### Ancillary changes.
Give reported program descriptive name of ODECS
Also report the program name and version at top.
Then report date and time of build.

Move the reason for reset below

LokiMetaSmith, 2023-09-19:
current output for spi when pulling one thermocouple 

`stage2Heater   : Ext2
Machine State: Off
Target    C: 30.00
Setpoint    C: 30.00
Temp        C: 28.00
Ramp    C/min: 0.50
Heater     DC: 0.0000
Bad post_stack_C
Current Heater to command:Ext1
Enter 1, 2, or 3 to switch the machine your commands change.
Use h:700 to set target temp; use r:0.3 to set ramp rate.
Bad post_stack_C
Bad post_stack_C
Current Heater to command:Ext1
Enter 1, 2, or 3 to switch the machine your commands change.
Use h:700 to set target temp; use r:0.3 to set ramp rate.
Bad post_stack_C
Bad post_stack_C
Current Heater to command:Ext1
Enter 1, 2, or 3 to switch the machine your commands change.
Use h:700 to set target temp; use r:0.3 to set ramp rate.
Bad post_stack_C
Stage2Heater   : Int1
Machine State: Off
Target    C: 30.00
Setpoint    C: 30.00
Temp        C: 27.50
Ramp    C/min: 0.50
Heater     DC: 0.0000
Stage2Heater   : Ext1
Machine State: NormalOperation
Target    C: 50.00
Setpoint    C: 50.00
Temp        C: 51.00
Ramp    C/min: 1.00
Heater     DC: 0.9750
Stage2Heater   : Ext2
Machine State: Off
Target    C: 30.00
Setpoint    C: 30.00
Temp        C: 28.00
Ramp    C/min: 0.50
Heater     DC: 0.0000
Bad post_stack_C
Current Heater to command:Ext1
Enter 1, 2, or 3 to switch the machine your commands change.
Use h:700 to set target temp; use r:0.3 to set ramp rate.
Bad post_stack_C
Bad post_stack_C
Current Heater to command:Ext1
Enter 1, 2, or 3 to switch the machine your commands change.
Use h:700 to set target temp; use r:0.3 to set ramp rate.`

ForrestErickson, 2023-09-20:
Notes:
The code, Rev: 0.3.5",  which does this is in the file "read_temps_task.cpp"
The function void ReadTempsTask::updateTemperatures() is currently on line 131.
The code which reports the totals is in the condition: if (DEBUG_READ_TEMPS > 0) {
which is currently at line 202. 
This block of code currently, 20230920 reports the one wire counts. SPI thermocouples are not reported.  

What appears to be the corresponding function for the stage 2 system is much less developed for reporting.
void stage2_ReadTempsTask::updateTemperatures() {
It is currently at line 224.

RobertLRead, 2023-09-28:
> @RobertLRead <https://github.com/RobertLRead>
> @LokiMetaSmith <https://github.com/LokiMetaSmith>
> Zero Thermocouple Amplifier Error Rate
>
> I have been running an Arduino sketch which is based on the library
> MAX31850_LIB_VERSION: 0.1.2 which reports the thermocouple fault types.
> *MAX31850_test_disconnect.ino*
> MAX31850_Test_Faults.zip
>
> As I have it now, the program is counting and reporting the four
> measurement types and unknown:
>
>    - VALID,
>    - ERR_SHORT_OPEN,
>    - ERR_SHORT_GND,
>    - ERR_SHORT_VDD
>    - NotKnown_Count
>
> I started it in the wee hours this morning and here is a journal with time
> stamps
> [image: image]
> <https://user-images.githubusercontent.com/5836181/271283083-c07760a7-a9af-4bea-ac1a-af41b4b7a332.png>
> Results
>
> In about 7 hours, I have run over 270K measurements with out error.
> [image: image]
> <https://user-images.githubusercontent.com/5836181/271281389-5a336c72-b9aa-4741-9676-617fc0655b77.png>
> Hardware Notes, and Test Limitations
>
> Mockup Of Maryville
> Two MAX81350 wired up however this program only reads one. I have not
> found an example with this library for reading multiple thermocouple
> amplifiers.

Good work!

When we get back and have rested, we need to think about big priorities.
This could be one of them.

ForrestErickson, 2023-10-04:
@RobertLRead 
@LokiMetaSmith 

I am going to have to interrupt this long duration test using the MAX31850_LIB_VERSION: 0.1.2  library and wanted to capture the status.

Over 2.5 million thermocouple reads with out reported error.
[image]
TestFaults20231001_0750.log 
Started about three days ago.

ForrestErickson, 2023-10-04:
## 20231004 Start experiment. Inject AC coupled 60 Hz interference into thermocouple.
Trying to find a way to duplicate thermocouple interference measurements of the ODECS.

### Interfering setup
Used a sheet of aluminum (about 9x11) foil folded in half to envelope the thermocouple coil. 
Use Siglent SDG830 to create 60 Hz interference. Set for 10Vp-p and couple output to AL foil and GND at the solderless breadboard.
HOWEVER, Still testing with the sketch using the MAX31850_LIB_VERSION: 0.1. library.

[image]
CH1 is thermocouple +, CH2 is thermocouple - and CH3 is the driven aluminum foil.

## Quick results
After 5K measurements no errors.
[image]

ForrestErickson, 2023-10-04:
## 20231004  . Inject AC coupled 60 Hz interference into thermocouple. DallasTemperature library V3.9.0 .
Trying to find a way to duplicate thermocouple interference measurements of the ODECS.

### Interfering setup
Used a sheet of aluminum (about 9x11) foil folded in half to envelope the thermocouple coil. 
Use Siglent SDG830 to create 60 Hz interference. Set for 10Vp-p and couple output to AL foil and GND at the solderless breadboard.
Firmware: testing with the sketch using the Miles Burton DallasTemperature 3.9.9 library 
[image]

Interfering signal.
[image]
CH1 is thermocouple +, CH2 is thermocouple - and CH3 is the driven aluminum foil.

## Quick results
After about 500 measurements no errors.
[image]

After 3K measurements no errors.
[image]

ForrestErickson, 2023-10-04:
Note previous waveforms mistakenly had a 3000 Hz function generator signal.
Here is the setup with function generator at 60 Hz.

[image]

ForrestErickson, 2023-10-04:
## Similar Hardware Test but with  milesburton/DallasTemperature@^3.11.0 Library
Lee loaded the Dallas_Tester_Rob project into VSC and Platform IO 
Set for the library: milesburton/DallasTemperature@^3.11.0

### Hardware note.
Turn off the function generator.
Remove the aluminum foil shield.
Remove the 0.1uF from the MAX31850 + and - input.
Program the Due.
Start running about 17: 15 EDT.

### Quick results.
No errors on the two amplifiers.
[image]
Note the pumpkin colored trace at 0 is the count / ratio of errors.

CC: @RobertLRead , @LokiMetaSmith

ForrestErickson, 2023-10-05:
## 20231005 . Inject AC coupled 60 Hz interference into thermocouple through 0.1uF capacitor. 
DallasTemperature library milesburton/DallasTemperature@^3.11.0
Trying to find a way to duplicate thermocouple interference measurements of the ODECS.

### Interfering setup
Direct couple through 0.1 uF into thermocouple tip.
Use Siglent SDG830 to create 60 Hz interference. Set for 1.0Vp-p and GND at the solderless breadboard.
Firmware: testing with the sketch using the milesburton/DallasTemperature@^3.11.0 library

### Quick Results at 1Vp-p
[image]
CH1 MAX31850 + input, CH2 MAX31850 - input,, CH3 Signal generator output.
Note during the assumed conversation period pulse the  60 cycle waveform becomes super imposed on the pulse.

At 1Vp-p amplitude the error rate is zero.
[image]

### Quick Results at 2Vp-p
Errors reproduced.
[image]
CH1 MAX31850 + input, CH2 MAX31850 - input,, CH3 Signal generator output.
Note during the assumed conversation period pulse the  60 cycle waveform becomes super imposed on the pulse.

Switch from 1Vp-p to 2Vp-p about mid graph.
At 2Vp-p amplitude the error rate is steadily climes.
[image]

Longer term graph same setup
[image]

### Quick Results at 1.65Vp-p
Approximate threshold of errors reproduced.
[image]
2Vp-p making lots of errors.
1.5V making no errors.
1.65 making intermittent errors.

### Conclusion
Injection of AC coupled 60 Hz can produce errors.

CC: @RobertLRead @LokiMetaSmith

ForrestErickson, 2023-10-05:
## 20231005 . Inject AC coupled 60 Hz interference into thermocouple through 0.1uF capacitor. 
DallasTemperature library milesburton/DallasTemperature@^3.9.0 (Same as Arduino IDE Version)

### Interfering setup
Direct couple through 0.1 uF into thermocouple tip.
Use Siglent SDG830 to create 60 Hz interference. Set for 1.0Vp-p and GND at the solderless breadboard.
Firmware: testing with the sketch using the milesburton/DallasTemperature@3.9.0

### Quick Results at 1.65Vp-p
CH1 MAX31850 + input, CH2 MAX31850 - input,, CH3 Signal generator output.
Note during the assumed conversation period pulse the  60 cycle waveform becomes super imposed on the pulse.
[image]
At this amplitude the error rate is zero.

### Increase Interference to 3.0V
Set generator to 3.0Vp-p
[image]

Measured error rate is still zero
[image]
Note however that the blue trace, TC0 the thermocouple into which the interfering signal is injected has an increased dither.

Switch to 2 and then 1Vp-p interference
[image]

ForrestErickson, 2023-10-05:
### Library Differences
Sample rate for the two different version in oscilloscope captures above
milesburton/DallasTemperature@3.9.0  is about 75mS between samples.
milesburton/DallasTemperature@^3.11.0"   is about 125mS between samples on some screen shots but not all.

### Lets double check Sample Rate
#### Set for milesburton/DallasTemperature @ 3.9.0 and compile
Processing due (platform: atmelsam; board: due; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Removing unused dependencies...
Library Manager: Removing DallasTemperature @ 3.11.0
Library Manager: Removing dependencies...
Library Manager: Removing OneWire @ 2.3.7
Library Manager: OneWire@2.3.7 has been removed!    
Library Manager: DallasTemperature@3.11.0 has been removed!
Library Manager: Installing paulstoffregen/OneWire @ ^2.3.7
Unpacking  [####################################]  100%
Library Manager: OneWire@2.3.7 has been installed!
Library Manager: Installing milesburton/DallasTemperature @ 3.9.0
Unpacking  [####################################]  100%
Library Manager: DallasTemperature@3.9.0 has been installed!
Library Manager: Resolving dependencies...
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM (8.2.0) > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash

I see on the oscilloscope both sample periods of 75 and 125 when the scope is set for 20mS/div.

#### Set for milesburton/DallasTemperature@^3.11.0
Removing unused dependencies...
Library Manager: Removing DallasTemperature @ 3.9.0
Library Manager: Removing dependencies...
Library Manager: Removing OneWire @ 2.3.7
Library Manager: OneWire@2.3.7 has been removed!
Library Manager: DallasTemperature@3.9.0 has been removed!
Library Manager: Installing paulstoffregen/OneWire @ ^2.3.7
Unpacking  [####################################]  100%
Library Manager: OneWire@2.3.7 has been installed!
Library Manager: Installing milesburton/DallasTemperature @ ^3.11.0
Unpacking  [####################################]  100%
Library Manager: DallasTemperature@3.11.0 has been installed!
Library Manager: Resolving dependencies...

I see on the oscilloscope both sample periods of 75 and 125 when the scope is set for 20mS/div.

So there is NOT a difference in sample time between the two libraries.  The different oscilliscope captures above was just an artifact of Lee setting the scope differently.

## Conclusion 
What ever the difference in the vulnerability to interference it is not because of a different sample rate as compiled with the test program.

ForrestErickson, 2023-10-05:
## Examine the Differential Signal with Oscilloscope Math
Use math to display the difference of CH1-CH2. 

### Set for signal generator at 0.00Vp-p and couple with smaller capacitor.
[image]
CH1 is thermocouple +, CH2 is thermocouple - and CH3 is the driven coupling capacitor 2x4700pF, Math is ch1-ch2
Nice rectangular samples and the math channel too.
[image]
Error rate is zero. 

### Set for signal generator at 12.00Vp-p and couple with smaller capacitor.
[image]
CH1 is thermocouple +, CH2 is thermocouple - and CH3 is the driven coupling capacitor 2x4700pF, Math is ch1-ch2

Error rate is zero.
[image]

### Set for signal generator at 13.00Vp-p and couple with smaller capacitor.
[image]
CH1 is thermocouple +, CH2 is thermocouple - and CH3 is the driven coupling capacitor 2x4700pF, Math is ch1-ch2

[image]
Error rate becomes non zero.

## Conclusion on Differential Signal
With the oscilloscope Lee cannot see why the error rate has a threshold.

ForrestErickson, 2023-10-05:
Waveform with + Terminal Disconnected.

[image]

RobertLRead, 2023-10-05:
Lee and I have studied the driver and observe that the code reports EVERY problem as a disconnect. That is, if the wires are tied to GND, tied to VCC, or actually disconnected, it is reported as a disconnnect. This a weakness in the driver code.

Worse, it also obscures electrical problems that would cause the CRC on the bytes under the same (DEVICE_DISCONNET). We therefore propose to fork that repo and hack the driver in our test program in an attempt to understand how it is that the a strongish AC coupling is causing a disconnect to be reported.

Lee has evidence on the scope of a sufficiently strong AC signal corrupting the "square waves" which occur under other circumstances. However, we do not yet know exactly how the bits reported back by the MAX31850 chip are treating this.

RobertLRead, 2023-10-05:
Dear @ForrestErickson,
I have now made the fork and connected it in Dallas_Tester_Rob. It shows that the driver is correctly finding an open condition and reporting it when a thermocouple is unplugged. However, as we have determined, it is PROBABLY also reporting EMI induced problems as disconnects. A weakness of this approach is that it polutes the Serial stream with lots of debugging statements; it will make your plotting different. I can probably remove that and make it safe to use with plotting, but we should probably pair on that.

ForrestErickson, 2023-10-06:
@RobertLRead 
Regarding, "I have now made the fork and connected it in Dallas_Tester_Rob."
I have loaded this code. I modified it to reflect the MOM with only two thermocouples.

I have it reporting the following output for normal readings.
[image]

For thermocouple disconnected ie open
[image]

For  thermocouple grounded
[image]

For  thermocouple to VCC
[image]

For 1Wire tied low with 1K
[image]
Note the update rate for the data became slower when I loaded the 1 wire with the resistor.

For 1Wire interface open, Disconnect the 1Wire.
[image]

### Summary and Conclusion
The code reports the DEVICE_RAULT_OPEN_RAW correctly for a disconnected thermocouple

The code reports the fault but does not report the fault type for thermocouple grounded.

The code reports the fault but does not report the fault type for thermocouple connected to VCC.

The code does not reports the bus fault when a 1K is connected to the 1 wire. 

The code does not reports the bus fault when the 1Wire interface is opened.

RobertLRead, 2023-10-06:
I think we should see what happens when we do the capacitative AC coupling
you have demonstrated.

The key experiment is to figure out what happens in the code path in that
case.

ForrestErickson, 2023-10-06:
@RobertLRead ,
Regarding, "I think we should see what happens when we do the capacitative AC coupling
you have demonstrated."

This is what is reported.
[image]

RobertLRead, 2023-10-07:
Dear Lee, @ForrestErickson  at 4:54 Central time Oct. 7th, I pushed a change to our fork of the Dallas "Arduino-Temperature-Control-Library" that will print the actual bits which are relevant to the error we seem to be seeing. If you will please remove your use of the local copy, and simply do a "make", which removes the ".pio" directory forcing our fork to be re-read and make a screen capture of the failure as you have done in the comment right above this, we will have an important clue what is happening.

ForrestErickson, 2023-10-08:
@RobertLRead Regarding, "Dear Lee, @ForrestErickson at 4:54 Central time Oct. 7th, I ...."
Acknowledged.

### Additional information, Not Reporting Resolution.
Back on the 6th I was experimenting with the library and am puzzled.  I was trying to use some other programs from the example folder of the library using the Arduino IDE (I do not just now remember why as about two days have passed.)  I was not able to get them to compile with your library I had cloned locally until I realized I needed to replace in the IDE library folder the DallasTemprature folder the 3.9.0 with your library. That I did this morning and could compile. 

#### Multiple With Rob Library
I found that the library example "Multiple" as modified by me, with your library did not run through the end of setup()
[image]
Program Multiple with last line of setup() highlighted
[image]

#### Multiple With library 3.9.0
I found the same program (only modified with the text indicating the library type) completed setup() and went into loop()
[image]
HOWEVER note that the resolution was reported as "0".

#### Dallas_Tester_Rob with Rob Library
I changed the Dallas_Tester_Rob code so as to report the resolution and found that this code too does not report the resolution .
[image]

ForrestErickson, 2023-10-08:
@RobertLRead ,
Regarding, "If you will please remove your use of the local copy, and simply do a "make", which removes the ".pio" directory forcing our fork to be re-read and make a screen capture of the failure as you have done in the comment right above this, we will have an important clue what is happening."

Currently on the Dallas_Tester_Rob  I believe the library is selected by  "	${extra.lib_deps_external}"
[image]

So when I built this morning may have been using the new code you pushed on the 7th October.

**HOWEVER.** Looking at this build report for dependencies  "DallasTemperature @ 3.11.0+sha.c43a494" is still reported
[image]

I cannot tell what library I am using.  It reports as if the "milesburton/DallasTemperature@^3.11.0" yet the code in the VSC shows the link to external.  It appears my build has not gotten the new library.

#### Screen shot of serial monitor
[image]

ForrestErickson, 2023-10-08:
To give visibility to version I added this code in setup after the serial port begin

  Serial.print("DALLASTEMPLIBVERSION: ");
  Serial.println(DALLASTEMPLIBVERSION);

I compiled and ran the code and got this
[image]

This seams to show I am not running Rob's library.

BUT Waite!  Rob's library at https://github.com/PubInv/Arduino-Temperature-Control-Library/blob/master/DallasTemperature.h
reports version with "DALLASTEMPLIBVERSION "3.8.1""
[image]

### Update Version on PublicInvention
//#define DALLASTEMPLIBVERSION "3.8.1" // To be deprecated -> TODO remove in 4.0.0
#define DALLASTEMPLIBVERSION "3.99999_20231007" // PublicInvention work of 20231007

I performed a Clean, a Build and Upload  but the program still reports 3.8.1.
[image]

ForrestErickson, 2023-10-08:
Check of the latest code in the library at: https://github.com/milesburton/Arduino-Temperature-Control-Library
shows that the hard coded version is still going to report 3.8.1
[image]

I am feeling that everything is suspect now.

ForrestErickson, 2023-10-08:
@RobertLRead 

There is a pull request to the latest code in the library at: https://github.com/milesburton/Arduino-Temperature-Control-Library
It is regarding processing of error codes.
See: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/238

ForrestErickson, 2023-10-08:
After much head scratching.
I updated my local copy of Rob's Arduino-Temperature-Control-Library 
C:\Users\Public\Downloads\Arduino\PubInv\DallasTemprature\Arduino-Temperature-Control-Library

In my libraries folder with name,   C:\Users\Admin\Documents\Arduino\libraries
I made a back up of the folder "DallasTemperature" already there. I deleted the original folder. 
I made a copy of Rob's library in my libraries folder with name "DallasTemperature".  

I modified my "Multiple.ino" program to reflect the library name and checke that the program reported the librar as modified and expected as: "DALLASTEMPLIBVERSION: 3.99999_20231007"
YES I am working with Rob's library in the Arduino IDE.

### Testing Library with 20231007 Changes
Here are readings with Rob's modified library

> 13:08:54.245 -> V3 Multiple, DALLASTEMPLIBVERSION: 3.99999_2023100
> 13:08:54.245 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.245 -> 1, 1, 81, 81
> 13:08:54.292 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.292 -> 1, 1, 61, 61
> 13:08:54.386 -> DALLASTEMPLIBVERSION: 3.99999_20231007
> 13:08:54.386 -> Locating devices...Found 2 devices.
> 13:08:54.386 -> Parasite power is: OFF
> 13:08:54.433 -> Device 0 Address: 3B38B7940C4C6CB4
> 13:08:54.433 -> Device 1 Address: 3B5AB9940C4C0CDE
> 13:08:54.433 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.433 -> 1, 1, 81, 81
> 13:08:54.481 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.481 -> 1, 1, 81, 81
> 13:08:54.481 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.481 -> 1, 1, 61, 61
> 13:08:54.528 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.528 -> 1, 1, 81, 81
> 13:08:54.528 -> Device 0 Resolution: Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.528 -> 1, 1, 81, 81
> 13:08:54.528 -> 12
> 13:08:54.528 -> Device 1 Resolution: Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.528 -> 1, 1, 61, 61
> 13:08:54.528 -> 12
> 13:08:54.528 -> End of setup
> 13:08:54.528 -> Requesting temperatures...DONE
> 13:08:54.620 -> Device Address: 3B38B7940C4C6CB4 Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.620 -> 1, 1, 97, 97
> 13:08:54.620 -> Temp C: 22.75 Temp F: 72.95
> 13:08:54.620 -> Device Address: 3B5AB9940C4C0CDE Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:08:54.668 -> 1, 1, 197, 197
> 13:08:54.668 -> Temp C: 22.50 Temp F: 72.50
> 13:08:55.139 -> Requesting temperatures...

### Analysis
This code reports resolution of 0

ForrestErickson, 2023-10-08:
## Fault testing with code DALLASTEMPLIBVERSION: 3.99999_20231007

### Disconnect Thermocouple
I disconnected the thermocouple from TC0 and get "FaultDetected!" error message with the codes.
[image]

### Short Thermocouple to GND
I shorted to ground the thermocouple from TC0 and get "FaultDetected!" error message with the codes.
[image]

### Short Thermocouple to VCC  (Vdd)
I shorted to VCC the thermocouple from TC0 and get "FaultDetected!" error message with the codes.
[image]

### Open 1 Wire to TC1
I opened the 1 wire to TC1, TC0 still connected. 
[image]
Sample text of results:

> 13:28:26.830 -> Requesting temperatures...DONE
> 13:28:26.923 -> Device Address: 3B38B7940C4C6CB4 Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:28:26.923 -> 1, 1, 248, 248
> 13:28:26.923 -> Temp C: 23.75 Temp F: 74.75
> 13:28:26.923 -> Device Address: 3B5AB9940C4C0CDE Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:28:26.971 -> 1, 1, 201, 255
> 13:28:26.971 -> Error: Could not read temperature data
> 13:28:26.971 -> 
 
### Short1 Wire to GND 
[image]
Sample text of results:
> 13:33:07.046 -> Requesting temperatures...DONE
> 13:33:07.793 -> Device Address: 3B38B7940C4C6CB4 Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:33:07.793 -> 0, 1, 34, 29
> 13:33:07.793 -> Error: Could not read temperature data
> 13:33:07.793 -> 
> 13:33:07.793 -> Device Address: 3B38B7940C4C6CB4 Debug (b,!isAllZerors, CRC1, CRC2:) 
> 13:33:07.793 -> 0, 1, 34, 29
> 13:33:07.793 -> Error: Could not read temperature data
> 13:33:07.793 -> 

### Conclusions
The library correctly detects and identifies the error conditions for OPEN, SHORT TO GND and SHORT TO VDD (VCC).

A device which has lost the 1 Wire interface only gets the error message "Could not read temperature data"
The failure of a single device with an open connection and the failure of all devices with a shorted 1Wire report the same as "Error: Could not read temperature data"  To help with troubleshooting the code driver or application, should identify as specifically as possible the details of "Error: Could not read temperature data" for each expected device. 

CC: @RobertLRead

ForrestErickson, 2023-10-09:
## Testing Library DALLASTEMPLIBVERSION: 3.99999_2023100 with AC Interference
Summary. 60Hz AC Interference can report as short to ground.

#### Setup
Testing with program "Multiple" using library DALLASTEMPLIBVERSION: 3.99999_2023100
Inject 60 Hz interference through 4.7//4.7 nF (approximately 0.01 uF) into thermocouple. 
Function generator set for 20Vp-p
Capture serial with TeraTerminal to log files.

#### Results
**Waveforms at 60Hz**
[image]
CH1 MAX31850 input +, CH2 MAX31850 input -, CH3 Signal Generator output.
Note the interfering signals amplitude.

**Captured Data with Faults 60 Hz** 
[image]
Note that the faults were all to ground.  The search on " short to ground detected" produced the same number of hits as the search for "FaultDetected!"

**Watching Terminal For Faults and Capture Waveform**
[image]
The terminal reported a burst of faults as the interfering waveform brought the sample window below 200 mV.

### Conclusions for 60Hz tests
AC interference on thermocouple can cause Fault
The fault reported is to ground.
The coupled 60 Hz signal is of an amplitude driving the thermocouple voltage to with in 100 to 200 mV of ground. It is no where near the Vcc of 3.3V.

CC: 
@RobertLRead 
@LokiMetaSmith

ForrestErickson, 2023-10-09:
Log files of the measurements above with and with out time stamps.
These logs with 60Hz interference.
[DALLASTEMPLIBVERSION_3.99999_2023100_Inteference60Hz.zip](https://github.com/PubInv/NASA-COG/files/12845620/DALLASTEMPLIBVERSION_3.99999_2023100_Inteference60Hz.zip)

Arduino Sketch Multiple.ino V4 used for these captures above.
[Multiple_V4.zip](https://github.com/PubInv/NASA-COG/files/12845635/Multiple_V4.zip)

ForrestErickson, 2023-10-09:
@RobertLRead 
@LokiMetaSmith 

Lee's conclusion.  The DallasTemprature library needs to be fixed / extended to report the error types.

The library  MAX31850_LIB_VERSION: 0.1.2  may form a good guide for how to report the error types as it does so.  People writing code for one or the other can have minimal change.

The error reporting of the MAX31855 library should be investigated and implementation made the same or similar too.  

To my knowledge, as of 20231009, only Lawrence has MAX31855 with which to try to duplicate failures and characterize the MAX31855 and its library.

The MAX31850 1wire implementation we have is additionally incomplete in that we do not have (that I know of) a general solution for addressing and assigning to the thermocouple locations. The A0,A1, A2 and A3 lines are not read to assign the thermocouple location.

RobertLRead, 2023-10-09:
This is very good work, Lee. I think we have definitely peeled back a layer
of the onion. It would be especially gratifying if we could observe the
asserted problems at the Salt Lake City cite being the same thing.

I think we should make a Pull Request to fix the library with more detailed
error messages. I will add this to our bug our tracking system.

ForrestErickson, 2023-10-09:
@RobertLRead ,

Please read this pull request information: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/238

Here is the files changed with that pull request.
https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/238/files

CC: @LokiMetaSmith

RobertLRead, 2023-10-09:
There is a MAX31856 chip which supports 50Hz and 60 Hz filtering. 

I would like to know if that is relevant.  However, we are hamstrung by the fact that we can't put a scope on our our SPI 31855 chips on the stage2 heater, or if these are used in the alleged OTS PID controllers (would be nice to learn that). 
https://www.analog.com/media/en/technical-documentation/data-sheets/max31856.pdf

We could, in theory, purchase some of these and attempt to replicate the error of AC capacitive coupling that Lee has developed. It would be interesting to know if this makes a difference.
However, it is now apparent that we should select not just hardware, but also what has the most widely used software driver.
The Adafruit MAX31856 library is not yet heavily used: 
https://github.com/adafruit/Adafruit_MAX31856 - 21 stars
https://github.com/adafruit/Adafruit-MAX31855-library - 93 stars
https://github.com/milesburton/Arduino-Temperature-Control-Library - 918 starts

LokiMetaSmith, 2023-10-11:
I'm somewhat dubious on choosing based on stars alone, I want to know code quality and if someone is actively managing issues on the driver library. 

Regardless of the chip selected, I think we should plan on having pads for the ability to add passive filtering, DC-DC galvanic isolation, and support for TC shielding either into the main chassis or the analog front end.

ForrestErickson, 2023-10-11:
@LokiMetaSmith regarding, 
"I think we should plan on having pads for the ability to add passive filtering, DC-DC galvanic isolation, and support for TC shielding either into the main chassis or the analog front end." propose an exact circuit.

ForrestErickson, 2023-10-24:
@RobertLRead,
 
Enough time has passed I am only guessing.
This library has NOT been integrated into the OEDCS.
This library as it stands prints status which inteferes with evaluation of the normal library using the serial ploter feaure of the Arduino IDE as in so many of my examples above.

To get reliable reporting of the failure types the library needs to be "finished" and put into the OEDCS firmware.

**Please confirm my memory.**

Next step.
I believe upgrade to this library and improved reporting should be added to Task 2 because the HORIBLE problems we have having with thermocouple measurements might get more insight with this change. (Add an issue to incorporate improved PI version of DallasTemprature library)

I might be able to clobber the Serial.print() statements that make the modified library messy.
I might be able to incorporate it into the firmware of the OEDCS.
I assume you would be much faster.
Can you advise if and how I should proceed?  
Lee

CC: @LokiMetaSmith

ForrestErickson, 2023-11-10:
@RobertLRead 
This issue is not closed and my memory is becoming increasingly stale.

I think that there remains work to do on the library. 
The library is now reporting the additional error types.
**However,** it is also sending Serial.print messages which must be removed.
My head is swimming with respect to where to make changes. Where to put the files so that we can build for the PlatformIO.  IIRC I changed the name of the standard library on my Arduino IDE install and put the modified library in where the standard library had been.   At this moment I do not know enough or remember enough for how to do the same for the PlarformIO.

In short I do not know what to do next.

Lee

CC: @LokiMetaSmith

ForrestErickson, 2023-11-13:
### Summary of Insights on Interference Sensitivity of MAX31850
For the MAX31850 injecting an AC interference at 60 Hz through 0.1uF caused errors. The threshold for errors was when the function generator signals was greater than 1Vp-p and less than 2Vp-p.

At 60Hz the impedance of the capacitor is 1/(2pi*f*c) = 1/(6.28*60*0.0000001) = about 26K.
For 2V/26K this is an interfering current of 77uA. 

The observed amplitude of interfering signal at the thermocouple inputs was about the amplitude of the rectangular pulse observed and assumed to be when the measurement is made and is about 500mV.  
For 77uA to make a signal on the order of 500mV the common mode input impedance of the MAX31850 must be about 0.5/0.000077= 6.5K

### Conclusion on Shielding
Shielding of thermocouples to reduce common mode current to less than 77 uA should be at the threshold of preventing interference and bad measurements.

ForrestErickson, 2023-11-14:
20231114 Resume work on DallasTemperature Library.

Working with VSC and platform IO.
Modifed the platformio.ini file to call out the Public Invention fork of the DallasTemprature library. 

> [ribbonfishlibx]
> ; ; lib_deps =
> ; ; 	OneWire
> ; ;	DallasTemperature
> lib_deps_external =
> 	https://github.com/PubInv/Arduino-Temperature-Control-Library.git

Lee found he had to also comment out all environments except the Ribbonfish to commpile with our errors.

Modified MAX31850.cpp to report the library version number 
>   //Serial.print("Thermocouple library version: ");
>   Serial.print("Thermocouple library version: ");
>   Serial.println(DALLASTEMPLIBVERSION);

Now load into the target. Test target with Thermocouple disconnected from one of the three amplifiers

> 15:28:51.469 -> DEVICE_FAULT_OPEN_RAW returned!
> 15:28:51.469 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 15:28:51.469 -> 1, 1, 49, 49
> 15:28:51.469 -> FaultDetected!
> 15:28:51.469 -> 11111101
> 15:28:51.469 -> 1000001
> 15:28:51.469 -> DEVICE_FAULT_OPEN_RAW returned!
> 15:28:51.516 -> Debug (b,!isAllZerors, CRC1, CRC2:) 
> 15:28:51.516 -> 1, 1, 49, 49
> 15:28:51.516 -> FaultDetected!
> 15:28:51.516 -> 11111101
> 15:28:51.516 -> 1000001

So at this point the modified library is being called by the OEDCS firmware project.

CC: @LokiMetaSmith

ForrestErickson, 2023-11-14:
@LokiMetaSmith 

I found the function which tests for errors is:

/ reads scratchpad and returns fixed-point temperature, scaling factor 2^-7
int32_t DallasTemperature::calculateTemperature(const uint8_t* deviceAddress,
                                                uint8_t* scratchPad) {

### Poor Error Test Coverage 
As implemented this function will report one and only one error.  The chip may report more than one error.