PubInv/NASA-COG #278: set point is not waiting for actual temp to match before increasing the next temp with ramp

LokiMetaSmith, 2023-09-19:
[image]

the set point is increasing and not in lock step with the temp, such that the actual ramp rate is higher than it should be

LokiMetaSmith, 2023-09-19:
letting it run longer and it's not stabalizing,

15:07:15.069 -> Stage2Heater   : Int1
15:07:15.069 -> Machine State: Warmup
15:07:15.112 -> Target    C: 60.00
15:07:15.112 -> Setpoint    C: 42.75
15:07:15.112 -> Temp        C: 28.25
15:07:15.112 -> Ramp    C/min: 0.50
15:07:15.112 -> Heater     DC: 0.0000
15:07:15.112 -> Stage2Heater   : Ext1
15:07:15.112 -> Machine State: NormalOperation
15:07:15.112 -> Target    C: 50.00
15:07:15.112 -> Setpoint    C: 50.00
15:07:15.112 -> Temp        C: 54.25
15:07:15.112 -> Ramp    C/min: 0.50
15:07:15.112 -> Heater     DC: 1.0000
15:07:15.238 -> Stage2Heater   : Ext2
15:07:15.238 -> Machine State: NormalOperation
15:07:15.238 -> Target    C: 40.00
15:07:15.238 -> Setpoint    C: 40.00
15:07:15.238 -> Temp        C: 49.00
15:07:15.238 -> Ramp    C/min: 0.50
15:07:15.238 -> Heater     DC: 1.0000
15:07:24.960 -> Current Heater to command:Ext1

RobertLRead, 2023-09-19:
Let me investigate that here. How long has the temp been above the setpoint
without  the duty cycle decreasing?

LokiMetaSmith, 2023-09-19:
5:36:15.067 -> Stage2Heater   : Int1
15:36:15.067 -> Machine State: Warmup
15:36:15.067 -> Target    C: 60.00
15:36:15.067 -> Setpoint    C: 56.25
15:36:15.067 -> Temp        C: 27.25
15:36:15.067 -> Ramp    C/min: 0.50
15:36:15.067 -> Heater     DC: 0.0000
15:36:15.067 -> Stage2Heater   : Ext1
15:36:15.067 -> Machine State: NormalOperation
15:36:15.067 -> Target    C: 50.00
15:36:15.067 -> Setpoint    C: 50.00
15:36:15.067 -> Temp        C: 52.25
15:36:15.067 -> Ramp    C/min: 0.50
15:36:15.067 -> Heater     DC: 1.0000
15:36:15.238 -> Stage2Heater   : Ext2
15:36:15.238 -> Machine State: NormalOperation
15:36:15.238 -> Target    C: 40.00
15:36:15.238 -> Setpoint    C: 40.00
15:36:15.238 -> Temp        C: 47.75
15:36:15.238 -> Ramp    C/min: 0.50
15:36:15.238 -> Heater     DC: 1.0000

LokiMetaSmith, 2023-09-19:
more than 30 minutes

RobertLRead, 2023-09-19:
Ugh. That means its a bad bug.

ForrestErickson, 2023-09-20:
I updated my fork and pulled to my local clone this morning (30230920). I checked the log and noted: 

> commit 2c64ef6c5637e33d95d3170458f907fade997a0f
> Author: Robert L. Read <read.robert@gmail.com>
> Date:   Tue Sep 19 18:45:24 2023 -0500
> 
>     fixing terrible bug #278
> 

I also note that the version in main is where I think I left it, "#define VERSION "; Rev: 0.3.4"  //"
I would like to see the version bumped with there are changes so that I know with what I am testing with / working.

closed 2023-10-09 by RobertLRead