PubInv/NASA-COG #281: Bad thermocouple reading caused an emergency shutdown.

RobertLRead, 2023-09-20:
at 8:35 today, at 255.75, the system went into an emergency shutdwon due to a bad thermocouple read.

In response I am going to set the toleration time to 30 seconds.

Machine State: Warmup
Target     C: 400.00
Setpoint   C: 274.75
Target Ramp C: 0.50
Max Stack   A: 3.00
Max Stack   W: 30.00
Fan PWM      : 0.60
Post Heater C: 255.75
Post Getter C: 169.00
Post Stack  C: 225.25
Heater DC    : 1.00
Stack amps  A: 0.07
Stack watts W: 0.84
Stack volts V: 12.07
Stack ohms  O: 172.43
Fan RPM      : 2355.00
Enter s:1 to Turn On, s:0 to Turn Off.
Enter a:XX.X to set (a)mperage, (w)attage, (f)an speed (h)eater set p., and (r)amp rate.
returning fan speed: 0.60
fan speed, amperage
0.60 0.42
Free Memory: 20375
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Running ReadTemps

Machine State: Warmup
Target     C: 400.00
Setpoint   C: 274.75
Target Ramp C: 0.50
Max Stack   A: 3.00
Max Stack   W: 30.00
Fan PWM      : 0.60
Post Heater C: 255.75
Post Getter C: 169.75
Post Stack  C: 225.75
Heater DC    : 1.00
Stack amps  A: 0.07
Stack watts W: 0.84
Stack volts V: 12.07
Stack ohms  O: 172.43
Fan RPM      : 2328.00
Enter s:1 to Turn On, s:0 to Turn Off.
Enter a:XX.X to set (a)mperage, (w)attage, (f)an speed (h)eater set p., and (r)amp rate.
GOT EMERGENCY SHUTDOWN!
CHANGING STATE TO: OffUserAck

Free Memory: 20375
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Running ReadTemps

Machine State: OffUserAck
Target     C: 400.00
[... 391 lines cut ...]
Post Stack  C: 226.75
Heater DC    : 0.00
Stack amps  A: 0.00
Stack watts W: 0.84
Stack volts V: 7.06
Stack ohms  O:  N/A
Fan RPM      : 0.00
AN ERROR OCCURED. WILL NOT ENTER OFF STATE 
UNTIL ACKNOWLEDGED. ENTER A SINGLE 'a' TO ACKNOWLEDGE:
Enter s:1 to Turn On, s:0 to Turn Off.
Enter a:XX.X to set (a)mperage, (w)attage, (f)an speed (h)eater set p., and (r)amp rate.
GOT EMERGENCY SHUTDOWN!
Free Memory: 20375
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Error: Could not read temperature data: 0
Bad post_heater_C
Running ReadTemps

Machine State: OffUserAck
Target     C: 400.00

ForrestErickson, 2023-10-24:
@RobertLRead 

With the experience with Serial Number 2 left at at AmOx can this be closed?

RobertLRead, 2023-10-24:
Yes.