PubInv/NASA-COG #284: Detect Loss of AC (Power) Input, Take Appropriate Countermeasures For Restart

ForrestErickson, 2023-09-20:
We have added a battery to the ODECS to supply the Due to cover short term loss of AC power.
The Due could detect the loss of AC power if loss of, for example, the +24V power supply output were to be an event.
This would require a resistive divider and an input into the Due to detect a low going event. 

Possible values
A 10K into 1.5K from 24V will give 3.27 volts.  So would 20K into 3K.
Power dissipation is about 24^2/10K = 57mW so a 1/8 W resistor would work.  

Sensing 24V going low, the Due could take countermeasures.
The Due would "understand" why the power supply was not responding on the UART interface for example 

**A Concern:** 
Is there a problem with possible current from the Due TX pin into the power supply RX pin which might interfere with the power supply recovery from loss of power or damage the GPIO pins on either devices?

The Due could know how long power is out and might better anticipate how to control the system when power is restored.  

The power loss event and power loss duration might be stored and reported for the system history.

The system could know or guess for how long data was not sent through the UDP Ethernet path.  Write information to SD card to buffer for eventual send when Ethernet is restored.

At power loss, the system could deenergize relays so that when AC returns there may be a return to power state with an orderly sequence and minimal surge currents.

RobertLRead, 2023-10-11:
This may overloop another issue, but should be managed as part of this milestone.

RobertLRead, 2023-10-19:
I like what is being suggested here. I don't know if this hardware needs to go into the v1.1 hardware design, but we could build a tiny protoboard for it in anycase. I recommend that we make this a high priority for the "One Button Milestone".

@ForrestErickson, why don't you breadboard up or solder up a solution, and then implement draft code as you have outlined to at least detect the issue (reporting it seamlessly will require more work.) That would be a nice step to the core functions we know we need to implement in this task.

If nothing else seems obviously more pressing, please prioritize this issue.

RobertLRead, 2023-10-19:
Note: I see that you created a 24V_SENSE signal line already in v1.1 as part of the related "Detection Loss of Blower Rotation" issue. That makes this even easier! Please move forward with it.

ForrestErickson, 2023-10-24:
Progress with the closing of Issues: 
Detect Loss of Service Power, Hardware Component, Control V1.1 PCB #330
Detect Loss of Service Power, Firmware Component #335 (Lee has created a function in the COG Task which returns True if the inferred AC power (+24V) is good and False if bad.)

We are part way there.
Lee has no idea what to do at the OEDCS state level with this information. 

As long as power is off the OEDCS should put variables into a state where if power is restored immediately we minimized harm. Lee assume that means, 

1.  Blower OFF and set a system state. 
2.  Power supply to a low forward bias level (prevent reverse current in COG.) and set a system state. 
3.  Heaters off and set a system state. 
4.  Capture power loss time time to a log. A log which does not exist as of 20231024 but we could stub some how.

@RobertLRead 
Can you please define the cases and counter measures as you see them?

RobertLRead, 2023-11-17:
This is correct. You have defined the counter measures correctly:

1. Blower OFF and set a system state.
2. Power supply to a low forward bias level (prevent reverse current in COG.) and set a system state.
3. Heaters off and set a system state.
4. Capture power loss time time to a log. A log which does not exist as of 20231024 but we could stub some how.

I'm not sure the "cases" you ae asking about. This seems the correct action if you lose mains/grid power.