PubInv/NASA-COG #298: compile time issue with  - toolchain-gccarmnoneeabi

LokiMetaSmith, 2023-09-28:
When compiling the source code on a rPi, we needed to switch out the toolchain compiler to a different version that had support for the arm chipset. In doing so, we increased the revision of the compiler

rPi doesn't work
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)

linux pc works
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)

It is assumed that this is a bug in the firmware and the older ver(7.2.1) toolchain didn't get hung up on it. However, with ver(9.3.1) the compiled code hangs and the watchdog timer triggers.

gmulligan, 2023-10-19:
I have tried all of the last 8 versions of the compiler and just like Lawrence said, if you use a compiler that will run on an arm 64 pi then the code hangs right after:
  bool serialReportAdd = core.AddTask(&serialReportTask, &serialReportProperties);

just for sake of completeness I'm going to try to compile on a 32 bit pi and see what happens.

LokiMetaSmith, 2023-10-23:
Capturing email discussion: 

I recommend we drop requirements for rPi and use an x86 NUC

If we budget the cost of a rPi https://www.canakit.com/raspberry-pi-4-starter-kit.html

I would get this as it's $80, more powerful than most rPi's, has all required ports and networking requirements and has 128GB of storage
https://www.amazon.com/ATOPNUC-Computer-Desktop-Suppport-Ethernet/dp/B0BBMMYQNB/

ForrestErickson, 2023-10-24:
Lee urges that this issue be removed from the Task 2 milestone list as it is not a deliverable.

LokiMetaSmith, 2023-10-25:
Closing as we have a workaround

gmulligan, 2023-10-25:
This is still an issue.  Though we found a work-around.  i tried to compile the code with a newer version of the compile on the same machine, using version 1.70201 it compiles and runs "fine"...  using any newer version tickles the BUG and causes the code to watchdog reset 

SO we have to fix the buggy code

RobertLRead, 2023-10-25:
Well, if this is true it comes as a relief --- except for being short of
time, this is a better path forward anyway.  I suppose we should device who
will take this on---let's discuss in email to make sure we have a group
decision.

LokiMetaSmith, 2023-10-26:
If it helps, I've found that platform.io has extensive lint and inspection tools for code quality, memory usage, and more.
#332 has more info

ForrestErickson, 2023-10-28:
I would like to know more.