PubInv/NASA-COG #299: Problem limiting the amps and watts might produce less than 1 volt in forward bias..

RobertLRead, 2023-09-28:
There is a subtle problem that might arise when the stack is very hot. In this situation, the resistance is low.

If the limitation of the amps and watts are strictly followed and these are set to zero, this might create less than 1.0 volt of forward bias. 

We need to ask AmOx, but if this voltage needs to be maintained, we to compute our amperage limit to never go below one volt. Rob has written candidate code for this which he will put in branch called "1VoltLimit". This code has not been tested.

RobertLRead, 2023-09-28:
As part of this issue I recommend that we write a function to make the stacks in "minimal operating mode". This can be used in many places in the COG_TASK (4) to simplify the code. It should set both the amperage and the voltage.

ForrestErickson, 2023-10-24:
Change issue to Problem limiting the amps and watts might produce less than 1 volt in forward bias, Firmware, Task 2 to reflect the delivery.

Lee believes this is redundant to #306 where Lee has outlined a solution.

RobertLRead, 2023-11-17:
I agree. Closing this.

closed 2023-11-17 by RobertLRead