PubInv/NASA-COG #304: Add a separate shunt resistor on stack power to get accurate low-amperage measurments.

RobertLRead, 2023-09-28:
At AmOx while working with @tomsrobots, we observed that the amperage and wattage measurements at low amperage are very inaccurate. This is not entirely unexpected, but is troublesome.

AmOx suggest adding to the V2 OEDCS a 1 milliohm shunt resistor and measuring that digital as a verification of the digital amperage system.

Note: At present, it is not entirely clear if the inaccuracy is a control inaccuracy or an inaccuracy in the digital reporting.

ForrestErickson, 2023-09-28:
A hall effect current sensor is probably better but I have no direct experience with them.  Lawrence and I speculated about the value of such sensors long ago but I do not remember if he indicated direct experience or not.

ForrestErickson, 2023-09-28:
Example of Hall effect sensors:
https://www.digikey.com/en/products/detail/allegro-microsystems/ACS781LLRTR-100U-T/6189099

https://www.digikey.com/en/products/detail/allegro-microsystems/ACS770LCB-100U-PFF-T/4473982