PubInv/NASA-COG #307: Danger to Q1 and Due if Blower Ground Open,

ForrestErickson, 2023-10-01:
If the ground wire to the blower were to open the FG and PWM pins into the Control v1 board may become 24V.
This might damage Q1 (check if rated for +24V) 
And the Due at the BLOWER_TACH pin.

RobertLRead, 2023-10-09:
@ForrestErickson what is an appropriate action or strategy here?

LokiMetaSmith, 2023-10-09:
Opto isolator between fan and DUE

ForrestErickson, 2023-10-09:
I think clamp zener diodes will suffice. We should induce the failure and make some direct measurements.

Another reason to build up identical systems and more of them.

ForrestErickson, 2023-10-11:
20231011 Make D1 and D2 into 5.1V Zener diodes to clamp the signals to safe values on Control V1.1.

ForrestErickson, 2023-10-24:
@LokiMetaSmith 
@RobertLRead 

Note that the diodes for this protection D1 and D2 were a bit larger than optimal for the foot prints on the PCB. Lee requested JLCPCB solder anyway and will have to inspect each diode on each of the 5 assemblies. There is another issue (Find part for D1 and D2 footprint compatible #368) regarding finding parts which are compatible with the footprint (or changing the foot print on a Control V2 PCB.

A "factory test " for D1 and D2 is to put a  1mA or so, current limited source into J20 pin 4 and J20 pin 3 and verify clamp at 5.1V the pins.

LokiMetaSmith, 2023-11-16:
moving this out of Task 2