PubInv/NASA-COG #317: Characterize Error Reporting of the MAX31855 Thermocouple Amplifier

ForrestErickson, 2023-10-09:
As reported in #273  

The error reporting of the MAX31855 library should be investigated and implementation made the same or similar too.

To Lee's knowledge, as of 20231009, only Lawrence has MAX31855 with which to try to duplicate failures and characterize the MAX31855 and its library.

ForrestErickson, 2023-10-09:
The library Adafruit-MAX31855 has example code folder.
In the example code is a sketch which checks and reports errors.
See:  https://github.com/adafruit/Adafruit-MAX31855-library/blob/master/examples/serialthermocouple/serialthermocouple.ino

The OEDCS code does not appear to be checking for errors. Nowhere in the project is their a hit on "MAX31855_FAULT_OPEN"

RobertLRead, 2023-10-09:
Thank you. Clearly we would work on making this code better, but it is quite possibly a dead letter. I'm not sure there is a path foward for this technology. The Stage2 heater machine may not lead anywhere, and even if it did, there is no reason to use the SPI thermocouples; we would like switch to I2C or simply use the OneWire approach.

ForrestErickson, 2023-11-17:
While trying to figure out how the MAX31850 efforts should be handled,  I visited the MAX31855 (SPI interface type) library written by Adafruit.
I wanted to understand more how errors were handled by that library.
That library has a separate function to return an error code as a type uint8_t.

Quite different from the pull request against the Milesburton library.