PubInv/NASA-COG #319: Research and order high-lead time components for Functional Gas Management Milestone

RobertLRead, 2023-10-09:
This is likely to be an O2 purity sensor and a pressure sensor.

John Graf has mentioned a "Paramagnetic O2 sensor". I don't know the specs on that. Common Scuba sensors read up to about 95%. Our specs for the functional gas management milestone state we need 99.0% purity test (that is, if you get less than that, it is an alert condition of some kind.)  The scuba-style sensors use a metal oxide of some kind and have a shelf-life of only one year; I think we need to do better than that. 

The gas management system shall include an oxygen analyzer, set to identify <99.0% oxygen as the
fault criteria for oxygen purity.

We need a pressure sensor that can go up to 200 psig, at about +-1 Psig (negotiable).

LokiMetaSmith, 2023-10-10:
Potential O2 sensor technology for consideration

Zirconium dioxide technology, Measuring 0.1-25% or 0.1-100% O2 and precalibrated
https://www.co2meter.com/blogs/news/how-does-a-zirconia-oxygen-sensor-work

Off the shelf item:
https://www.mouser.com/ProductDetail/SST-Sensing/O2S-FR-T2-18C?qs=3jNSNtuqJTLwFgfWcyuBvw%3D%3D

LokiMetaSmith, 2023-10-12:
This device looks like it could work for low pressure i2c sensing 

https://www.digikey.com/en/products/detail/amphenol-advanced-sensors/NPI-19J-300G2/15789400

LokiMetaSmith, 2023-10-19:
https://www.celcoinc.com/services/oxygen-cleaning/

some information about oxygen rated equipment and requirements.

LokiMetaSmith, 2023-10-19:
[image]
high lead time, high cost, high pressure rated

LokiMetaSmith, 2023-10-19:
another supply for i2c configured high pressure oxygen rated transducers rated for 5000 psi and above.

https://appmeas.co.uk/products/pressure-sensors/i2c-pressure-sensor/

ForrestErickson, 2023-10-19:
Looking at the APPMEAS device mechanical drawtings I note the following:
[image]

This device must be threaded into the high pressure O2 at a G1/4 female port.  I do not know what the "G 1/4" is.

Then an electrical cable terminated with M12 connector at the high pressure port comes to the OEDCS terminated in a TBD connector. 

Note the references to IP65 for the gland. I recognized IP65 as a "wash down" rating and a quick google gives

`IP64 Protected from total dust ingress. Protected from water spray from any direction. IP65 Protected from total dust ingress. Protected from low pressure water jets from any direction. IP66 Protected from total dust ingress.
`

We will need to know from where to purchase or how to make an I2C cable with these IP64  M12 gland nuts.

ForrestErickson, 2023-10-19:
Here is M12 x 4 ( four conductor) Panel mount to 0.2 Meter wires.
https://www.l-com.com/ethernet-m12-4-pin-a-code-male-receptacle-ip67-rated-rear-mounting-style-with-02m-leads?gclid=CjwKCAjwp8OpBhAFEiwAG7NaElDBC4y7qmnsc3CDAHEPGLRlweDas7c7qhhK1o380edCj8jtSQOSqxoCwaMQAvD_BwE

I2C pinout 
[image]

Here will be the pin out through the header
[image]

I have requested of the manufacturer a 3D model of the bulkhead connector. In the mean time I have created a simplified model with some guesses for dimensions and pushed it here:
https://github.com/PubInv/NASA-COG/blob/develop/mech/enclosure/M12BulkHead/M12BulkHeadToBlunt-PartPadWries.stl

Here it is embedded in the 2U drawing.
[image]

ForrestErickson, 2023-10-19:
Some notes on connector gender. The pressure sensor has male pins. The bulk head connector I (Lee) identified has male pins. We will need a female to female straight through cable between the sensor and the Front Panel connection.

LokiMetaSmith, 2023-10-20:
I've got inquiries in with Amphenol, Applied Measurements, and TE for more information.

LokiMetaSmith, 2023-10-20:
https://www.te.com/usa-en/product-20003318-00.html

This product family looks promising for low/medium pressure sensing

LokiMetaSmith, 2023-10-20:
https://www.te.com/usa-en/product-20010085-72.html

this one can do up to 5000psi for under $100

ForrestErickson, 2023-10-20:
Please request help locating 3D models of selected parts. 
I have begun making folders for critical parts in the path: NASA-COG\mech\enclosure

LokiMetaSmith, 2023-10-29:
email notes:

First off, for specifications, we would like to add a pressure gauge to the Stage 1 control board so we can add pressure regulation to the control algorithm. One issue is that I don't have a good specification for how precise the pressure readings need to be or the nominal operating pressure range. Nominally, we've selected something that can handle full range 0-350psi, but wanted to verify that. Same goes for the high pressure side, we aren't building anything for that as it's not in the contract, but I wanted to spec a pressure transducer for what will end up being a gas management system that's independent of the Stage 1 controller. Right now that's spec'd at 3500psi. As you probably know, gauge pressure and proof pressure are such that the pressure rating of the gauge is 2x the gauge rating.

Secondly, because we're dealing with oxygen, the datasheets are a bit cagey as to their suitability for pure oxygen. Since the oxygen setup was built by AmOx, I'm curious if you bothered with flushing the system for use with O2, what gaskets and greases you settled on, what tube diameter, fitting sizes, and misc. hardware was used for the Pathfinder test. I "think" we can just replace some o-rings on a standard stainless steel pressure transducer, but dealing with high purity O2 is new to me. They sell "oxygen rated" transducers, but options and availability are limited, and I suspect they're designed for direct integration and not for OEM builds that need to be flushed anyways. 

Third, We have a requirement for oxygen quality sensing. This is a bit tricky because something off the shelf and fit for purpose isn't really available and it's not clear if the high pressure side needs one as well as the low pressure side. I suspect that only the low pressure side needs to validate O2 purity as part of the pumping and flushing, but I could be wrong. 

The other issue is that the sensor we've identified as a candidate can't handle the pressure of either the high pressure side or the low pressure side. To get around that issue I wanted to know if a pressure regulator and a one way purge solenoid valve assembly would work. This would allow us to sample O2 purity on intervals without damaging the O2 sensor and allow for samples to be pulled from either the high pressure tank or the low pressure tank. I suspect there would need to be a flush between samples, but I expect that's acceptable. 

Right now, we're looking at a handful of pressure gauges:

The  M3200 family from TE has options for everything between 100psi to 10kpsi and has an easy to integrate i2c interface, so on paper it seems like a good fit.

We're looking at the  M32JM-00010B-350BG, which has a gauge range from 0-350 PSI for integration into the Stage 1 control board. Stage 2 would likewise have a similar configuration with an appropriately rated gauge.
https://www.mouser.com/ProductDetail/571-M32JM00010B350BG
[M3200 Datasheet](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FM3200%7FA19%7Fpdf%7FEnglish%7FENG_DS_M3200_A19.pdf%7F20003318-00)

TE also recommended the M5200 series, as they offer Oxygen Clean B40.1 Level-4, but it's an analog only transducer, they don't have any on hand, and I question if it's necessary if we replace the M3200 with appropriate o rings.
https://www.te.com/usa-en/product-CAT-PTT0013.html

We're also looking at https://appmeas.co.uk/ as an option, but they have a similar lack of available transducers with the required interface (i2c or spi for mcu communication)

For the O2 sensor, we're looking at a https://www.mouser.com/ProductDetail/SST-Sensing/ZrO2-System-485?qs=3jNSNtuqJTLI%2FbQpiEeOeA%3D%3D which is an SST Sensing zirconium dioxide o2 sensor, It has a max pressure rating of around 80psi (6bar) and a nominal operating pressure of 0-3bar. 
https://sstsensing.com/product/screwfit-zirconium-dioxide-oxygen-sensors/

The idea would be to put a simple oxygen regulator on the input (260psi max inlet pressure)
https://store.epsgroupllc.com/oxyreg-35-psi-npt-oxygen-regulator-234KM1235/

and something like this to vent and flush the O2 sensor.
https://www.mcmaster.com/1190N22/