PubInv/NASA-COG #321: Create a v2 OEDCS control board to support independent work and the Functional Gas Managment

RobertLRead, 2023-10-09:
This issue is not about doing the gas management. I anticipate that we will need a replicatable v2 OEDCS control boards to allow for geographically distributed tested. We know there is rework on the v1 board. This issue is to repair that and to design and order a board that would allow for the delivery of additional 2-stack OEDCSs. 

I anticipate that we need 4 of these for our team, and up to two by early next year.

ForrestErickson, 2023-10-10:
On 20231010 Lee created new folders for version 1.1 as: Control_1v1\Control_1v1

ForrestErickson, 2023-10-11:
Graceful shutdown button on the board added.
S2 to D49.

[image]

ForrestErickson, 2023-10-13:
PCB V1.1 Board nearing completion
[image]
No DRC Errors. Plenty of warnings about silk screen clearances as expected.
[image]

Generating files, Gerbers and Drill and Placement.

Finding some discrepancies on the Headers / Sockets. Will have to resolve tomorrow. Not enough brain cells tonight.