PubInv/NASA-COG #323: Detect Blower Loss of Power

RobertLRead, 2023-10-10:
This would likely be checked in the "_run" routine in the CogTask class. (line 81, cog_task.cpp)

If the blower RPMS do not match the blower power, and if this has been true for more than about 5 seconds, we should raise this as an alert. We are not currently storing the history of the RPMS. Possibly a simple way to handle this is to perform this check if it has been 5 seconds since the most recent fan speed change. The time of the most recent fan speed changed is NOT currently stored anywhere, that would have to be added. 

At first glance, I would think the appropriate repsonse is an "automatic shutdown" ( that is, simply setting
ms = EmergencyShutdown). 

This should also be reported by writing to the Serial output.

However, we need a better developed error log internally; we have written some code for this but it is not yet well developed. We can consider proper logging of fault events a different issue, not part of this particular issue.

What constitues "too low" would need to be defined. We could construct a map or algebraic relationship showing the expected RPMS at each 10% change in the fan speed (from 0 to 100%) and throw the exception if we are at only 50% of that. That might be too sensitive.

A simpler test less subject to false alarms would be:
if the power is about 20% and the RPMS are less than 200.

Note: If you are careful, you can stop the fan with your finger, or better with a gloved finger, but applying pressure to the center of the exposed spindle. It may burn your finger if you do this at high speeds but it works. Be sure not to touch the impeller blades at the edges.

ForrestErickson, 2023-10-11:
@RobertLRead 

On Control V1.1 Lee has added a resistive divider on the +24V to make a signal "SENSE_24V" into the Due A1 input. 
The divider ratio is 4.7/34.7. = 0.1354. So 24V will be 3.25V.

The Due will need to pole this ADC input.

CC: @LokiMetaSmith

RobertLRead, 2023-10-19:
The hardware feature above is great, but it is very close to issue #284 .

I will rename this issue to be "Detect Blower Loss of Flow". However, if v1.1. is already done, I be #284 can now be undertaken directly (I think @ForrestErickson can at least start that code to the level of detection.)

ForrestErickson, 2023-10-24:
Given the name of this issue Lee thinks the solution must be a firmware function based on the tachometer and some function of the set PWM. 
Based on the experience writing the Detect Loss of Service Power, Firmware Component #335 this will be a function in the COG Task files.
Lee has renamed this issue to include "Firmware".

ForrestErickson, 2023-10-24:
### Blower RPM Versus PWM
Lee recorded a curve for the FG (Tachometer) versus PWM setting on his github at this repository:
https://github.com/ForrestErickson/FanControlLinearized
HOWEVER, this graph is probably not the true, raw, PWM setting because IIRC, the code has a map function which removed some of the low PWM settings. 

See the "[The linearizing function](https://github.com/ForrestErickson/FanControlLinearized#the-linearizing-function)" there.

Regardless here is a screen shot:
[image]

@RobertLRead 
Lee assumes you did not implement the linearizing function. Can you confirm?  

### Implication of No Linerization
The consequences of not linearizing is to perhaps make the control loop stability different in some regimes of operation.  Low RPMs will have higher gain and instability may be found there.

RobertLRead, 2023-11-17:
I have not linearized anything.