PubInv/NASA-COG #326: Detect Heater Loss of Power

RobertLRead, 2023-10-10:
The Heater Losss of Power is a fairly likely event, in that a the heater wires are not expected to have an infinite life. If a wire breaks, the heater circuit will be open.

Detecting that the circuit is open when you intend it to have a load would be ideal. However, detecting it from the heat signature should be possible, and it would be obvious to a human observer.  A good logical condition for it is a bit trickier. However, if you have one of fully functional OEDCSs it is easy to simulate; just unplug the extension cord which is connected to the heater.

Note that the "post_heater_C" temperature is read very close to this heater, and is (usually) the fastest to react to a temperature change.  If it fails to increase for a period of time as the heater is cycled on, this likely means the heater is not receiving power.

Under operation, this condition could be tricky to find, as the stack may be generating sufficient heat that the duty cycle on the heater is zero!

If you were willing to interrupt normal operation, the following procedure should work:
1. Turn the heater off for 10 seconds.
2. Record post_heater_C
3. Turn the heater on full power for 10 seconds.
If the post_heater_C is not higher than the recorded temperature, the heater is surely not getting power.

However, that procedure interferes with normal operation; it should only be used in extremis.

If the heater loses power, we are unlikely to be able to hold to ramp schedule or possibly to maintain temperature at normal operation. However, those conditions should raise there own alarms. It is possible that this condition should not be tested above 300 C, and we should rely on other tests to detect a problem.

Below 300C, if we fail to meet a ramp up or ramp down rate (by being too cool, in either case), we could execute the 20-second test above as a definite condition.

Note: A task must "pet the dog" every 16 seconds. If the test above is written as an explicit test, it would have to carefully take that into account.

ForrestErickson, 2023-10-10:
Regarding, "Detect Heater Loss of Power"  
Interesting As stated I would have assumed this meant detect loss of AC into the heater and would not have interpreted this to mean detecting a failure (open) of the heater element.  

Here are some inductive current transductors: https://www.sparkfun.com/products/11005 but I do not see an amplifier board for 15A or 20A.

ForrestErickson, 2023-10-24:
Lee renamed to "Detect Heater Loss of Power (Loss of Control), Firmware, Task 2" to better reflect this Task 2 issue. 
Given that our Serial Number 1 and 2 hardware and the Control V1.1 assemblies have no way to sense current through the heater it is only by the above speculated method, loss of change of temperature with change in SSR control that we might detect loss of heater control. 

@RobertLRead regarding, 
> If you were willing to interrupt normal operation, the following procedure should work:
> Turn the heater off for 10 seconds.
> Record post_heater_C
> Turn the heater on full power for 10 seconds.
> If the post_heater_C is not higher than the recorded temperature, the heater is surely not getting power."
is this likely to be code spread across several tasks and task libraries?  
Any suggestions?

LokiMetaSmith, 2023-11-16:
At one point we had talked about measuring the ramp rate vs the expected ramp rate, and throwing an error when the ramp rate is outside of bounds. 

This makes assumptions based on the fan speed, the heater PID setting, and the target temperature, and so would take all of those into account.