PubInv/NASA-COG #328: Detect DC loss of power

RobertLRead, 2023-10-10:
This can have various interpretations, but the most obvious approach is to tell if we can't put power into this stack. This is most likely to occur because we have lost communication with the digital power supply, usually because it has no AC power.

The simplest version of this task is to detect when we have lost communication with the digital power supply.

I suspect Geoff's code provides return values that can be checked but are currently being ignored. I think this should be done in "_updateCOGSpecificComponents" in cog_task.  If communications has been lost, it will likely be detectable here in a return value, (if they are being passed on.)

The current routine for updateSpecificComponents is of type void. It would probably be best to have it return an error condition. At present, that could just be a boolean, and the calling function could adjust the MachineState to EmergencyShutdown.

RobertLRead, 2023-10-17:
When an alert is thrown, the basic plan is:
1) To change state to either "CriticalFault" or EmerencyShutdown.
2) State Changes should be logged as Events (but they are not now.)
3) We have not produced a disiplined Event logging yes.

(A preliminary solution is just to log to the Serial port.)

ForrestErickson, 2023-10-24:
Lee renamed to better reflect issue to, "Detect Loss of Control of  DC Power to Stack, Firmware, Task 2"
Adding Lee and Robert to the assignee for visibility on this.