PubInv/NASA-COG #331: Design and Build an exceptional event logging system

RobertLRead, 2023-10-10:
This may be considered and enhancement.

I think we should implement a simple in-memory list of error conditions and the time at which they occur. This can be a simple ring-buffer of say 10 or 20 items.

When something enters the ring buffer, we should log it on the serial port and also the Network Interface. Geoff says this is almost ready---but we should document it all so that it can be handled by Cumulocity correctly. 

Ben Coombs wrote a primitive system for this into the core functionaly, but we have never used it. I think we should consider doing this as a part of the core.

RobertLRead, 2023-10-11:
This is subsumed by other issues.

ForrestErickson, 2023-10-24:
Lee added all team members to this Task 2 deliverable.
Changed name to reflect scope and delivery.
Design and Build an exceptional event logging system, Firmware, Task 2