PubInv/NASA-COG #332: Create a system for internally logging the last 10 minutes of critical parameters

RobertLRead, 2023-10-11:
We need to develop a system of logging all the main parameters (those which are already reported) for the last 10 minutes at the rate of 1 record per second. 

This is 600 logs of about 10 parameters; it should not be a memory burden. It should of course be implemented as a ring buffer which is statically allocated in memory.

When a critical fault or an emergency shutdown occurs, we want to dump this 10 minutes of records into the logs (the Network logs and the Serial log). This should allow fine-grained post-mortem analysis. 

I believe the current code and "reporting" mechanism will support this nicely. However, we will definitely have to make sure that reporting is done a separate schedule that we can control. We need to write to the in-memory log once per second, rain or shine.

ForrestErickson, 2023-10-24:
Rename to reflect status as a firmware Task 2.

This task is dependent on #331

LokiMetaSmith, 2023-10-25:
finished initial cut of this commit on  the branch log_recorder

it appears that creating 600 records of MachineStatusReport which has 15 floats and 1 long was too memory intensive

https://forum.arduino.cc/t/compile-cpp-elf-section-bss-is-not-within-region-ram/259981

LokiMetaSmith, 2023-10-25:
So, the issue is a classic time series compression problem, where we have a series of variables that vary over time but they can both vary wildly or not at all and we can't know before hand. 

https://www.timescale.com/blog/time-series-compression-algorithms-explained/

One problem is that we immediately convert all parameters to float, which is notoriously difficult to compress. 

I found one high quality library that appears to have the appropriate functions for running a time series compression algorithm, without having to write a library ourselves. 
https://www.arduino.cc/reference/en/libraries/serie/

Another option would be to refactor all of the code to use integers as opposed to floats, however I don't see this as practical

I'll create a test function that creates a moving window of real time logs, "say 10-30" and then compress those into a polynomial representation for storage. There will be artefacts in any compression algorithm, however I consider that acceptable in this use case.

LokiMetaSmith, 2023-10-25:
96KB SRAM, 512KB flash
 and using an estimated 26KB of ram for the current buffer of 600 records with each record including 16 entries 4Bytes each

LokiMetaSmith, 2023-10-26:
Before refactoring 
RAM:   [========= ]  93.9% (used 92328 bytes from 98304 bytes)
Flash: [==        ]  16.7% (used 87780 bytes from 524288 bytes)

Using arduino F macro

RAM:   [========= ]  93.9% (used 92328 bytes from 98304 bytes)
Flash: [==        ]  16.8% (used 87836 bytes from 524288 bytes)

Literally no difference, and over an hour's worth of refactoring.

RobertLRead, 2023-10-26:
??? What refactoring did you do?

LokiMetaSmith, 2023-10-26:
So, the CLI pio has a weird and wonderful feature. 

Type "pio home" and it starts a locally hosted web server and opens up all of the development tools usually found in the vscode platformio tool.

On the home tab, go to "Open Project" and open up the NASA-COG firmware directory where platformio.ini resides
[image]

Open up the Inspect tab, and select the environment your building against.
click "Inspect" and wait for the results
[image]

And clarity!

[image]

NOTE: these are the largest objects in the program
64 KB packetBufferpacketBuffer
20.1 KB machineConfigmachineConfig
5.2 KB _svfprintf_r_svfprintf_r
3.8 KB _strtod_l_strtod_l
3.5 KB _dtoa_r_dtoa_r

LokiMetaSmith, 2023-10-26:
in network_udp.cpp, this buffer consumes 2/3 of all available ram
// buffers for receiving and sending data
#define buffMax 64*1024

LokiMetaSmith, 2023-10-26:
changing 
#define buffMax 64*1024  
to 
#define buffMax 32*1024  

in network_udp.cpp,

and changing 
MAX_RECORDS = 300;
to 
MAX_RECORDS = 600;
in machine.h

should resolve the issue. 
@gmulligan can you comment on how much the buffMax buffer can be reduced while not affecting functionality?

LokiMetaSmith, 2023-10-26:
Re: "??? What refactoring did you do?" @RobertLRead 
review the file diff and change log on the git branch "log_recorder"

LokiMetaSmith, 2023-11-15:
[#](https://github.com/PubInv/NASA-COG/tree/log_recorder_integration)

has further integrations of log recording

closed 2023-11-17 by RobertLRead