PubInv/NASA-COG #335: Detect Loss of Service Power, Firmware Component

ForrestErickson, 2023-10-11:
For Detect Loss of Service Power, Control V1.1 PCB #330 is in design.

Firmware to pole A3 and A4 and set thresholds and set alerts is required to finish out the enhancement.

ForrestErickson, 2023-10-20:
Update code, Lee on 20231020.

define VERSION "; Rev: 0.3.9"  // Adding power_monitor_task,  Loss Detection by +24V

Tested with MOM and 24V bench supply .

The function reports the supply as good or bad on the serial monitor.
The function returns "true" for good and "false" for bad.
Does this make any sense for a task if there is nothing to read it?

Anway here is the code

```
bool PowerMonitorTask::_run()
    {
      // Note:adding a task
       Serial.println("PowerMonitorTask run");

        //Analog read of the +24V expected about 3.25V at ADC input.
        // SENSE_24V on A1.
        // Full scale is 1023, ten bits for 3.3V.
        //30K into 4K7 
        const long R1=30000;
        const long R2=4700;
        const float Vcc = 3.3;

        Serial.print("analogRead(SENSE_24V)= ");
        Serial.println(analogRead(SENSE_24V) * ((Vcc * (R1+R2))/(1023.0 * R2))); 

        if (analogRead(A1) > lowThreshold24V) {
            powerIsGood = true;
            Serial.println("+24V power monitor reports good.");
            return true;
        }else{
            powerIsGood = false;
            Serial.println("+24V power monitor reports bad.");
            return false;
        }
    }

```

RobertLRead, 2023-10-20:
Good work. I suggest this become on of many checks in a "health_check". At
present, the obvious place for this is at the beginning of the _run
function of the CogTask. However, it could be created as a separate task if
necessary.

it could be added to the "report" by adding a variable to hold status or
events.  We should design an "event" system that can store 10-20 events in
a ring buffer and report them. However, you can leave that to me---having
detected the problem is valuable progress.

ForrestErickson, 2023-10-20:
Regarding, " store 10-20 events in a ring buffer and ". 
I am wondering if "circular buffer" is synonymous with "ring buffer"  I was exposed that the "circular buffer" vocabular with digital signal processing where input samples were added to and removed from a circular buffer. 

I note Arduino libraries for both names.
[image]

RingBuffer
https://www.arduino.cc/reference/en/libraries/ringbuffer/
This library allows to use ring buffer with and without interrupts.
RingBuffer is a library for ring buffers. Elements can be of arbitrate type.

CircularBuffer
https://www.arduino.cc/reference/en/libraries/circularbuffer/
Arduino circular buffer library
A flexible, compact (~350 bytes overhead) and template based library providing a circular buffer implementation supporting both LIFO and FIFO usage.

ForrestErickson, 2023-10-21:
# Stating An Assumption, +24V Means Mains OK
It might be time to state that "Service Power" means the AC Mains power into the OEDCS. The method implemented it to infer the presence of AC Mains power by measure of the +24V supply output.  Except in the case of failure of that supply, the presence of +24V indicates some Mains connection with in the range at which the +24V supply can regulate.

ForrestErickson, 2023-10-23:
Merged new code 20231023.
[ForrestErickson](https://github.com/PubInv/NASA-COG/commits?author=ForrestErickson) [Merge pull request](https://github.com/PubInv/NASA-COG/commit/8e2a53da899a4e497a3e1471d65805f01c6e009e) https://github.com/PubInv/NASA-COG/pull/375 [from ForrestErickson/develop](https://github.com/PubInv/NASA-COG/commit/8e2a53da899a4e497a3e1471d65805f01c6e009e)
…
[8e2a53d](https://github.com/PubInv/NASA-COG/commit/8e2a53da899a4e497a3e1471d65805f01c6e009e)
[1 minute ago](https://github.com/PubInv/NASA-COG/commit/8e2a53da899a4e497a3e1471d65805f01c6e009e)
Git stats
[ 1,022 commits](https://github.com/PubInv/NASA-COG/commits/develop)

ForrestErickson, 2023-10-23:
Stale task files and refences deleted. 
Closing as done.

ForrestErickson, 2023-10-24:
Closing as done.

closed 2023-10-24 by ForrestErickson