PubInv/NASA-COG #336: Non Unique Reference Designators on Control V1.1 Morning of 20231012

ForrestErickson, 2023-10-12:
Examining the schematic first page I see significant repetition of component reference designators.
Reference designators must be unique before work can begin on the PCB.

@LokiMetaSmith 
Please resolve the reference designators to be unique.  Ideally not changing any reference designators share with the V1 board.

### Schematic Morning of 20231012

[image]

### Schematic Evening of 20231011
This problem is not in the schematic as I pushed the branch last evening.
[image]

### Annotation Settings 
During development of a schematic I may from time to time change the settings for annotating reference designators but here are my settings just now:
[image]

LokiMetaSmith, 2023-10-12:
> Assigned #336 <https://github.com/PubInv/NASA-COG/issues/336> to

This wasn't an issue when I pushed it last night. I'm investigating but I'm
assuming at this point that you've introduced errors into your local repo.

[image: image.png]

ForrestErickson, 2023-10-12:
Note that there are 1Ohm resistors with a value of 1R and some with 1R0.
[image]

The resistor with the value of 1R I had eliminated from the parts list and standardized on the 1R0 resistor as the fields on it were the more complete.

Here is the list Part List from last night showing zero 1R resistors.
[image]

Please standardize on the 1R0 part.

ForrestErickson, 2023-10-12:
View with Kicanvas.

https://kicanvas.org/?github=https%3A%2F%2Fgithub.com%2FPubInv%2FNASA-COG%2Fblob%2Fdevelop%2Felec%2FControl%2520Board%2FControl_1v1%2FControl_1v1%2FControl_1v1.kicad_sch

LokiMetaSmith, 2023-10-12:
this is a rendering issue with kicanvas. I didn't change these and they
look fine on the repo
[image: image.png][image: image.png]

LokiMetaSmith, 2023-10-12:
the resistor reference values have been unified around 1R0
[image: image.png]

LokiMetaSmith, 2023-10-16:
closing as this is complete, I pushed a bug report to kicanvas

closed 2023-10-16 by LokiMetaSmith