PubInv/NASA-COG #340: RepRap Connector Needs Shroud.  Needs More Clearance

ForrestErickson, 2023-10-13:
The connector keep out for J35 and J36 RepRap interface does not have proper keepout.

[image]

Was footprint: Connector_PinHeader_2.54mm:PinHeader_2x05_P2.54mm_Vertical

Needs to be footprint:  Connector_IDC:IDC-Header_2x05_P2.54mm_Vertical
[image]

For J35 and J36, this footprint collides with the resistors and with each other.

ForrestErickson, 2023-10-13:
@LokiMetaSmith 
Please call.

I can get better PCB area utilization in a very tight space if I can reassign, reorder, the pins to the RepRap but I am not sure what is allowed.  Here are the resistors to which routing can be made with fewer cross overs.  
[image]

And here is where they are on the PCB. 
[image]
Notice that if I could route more directly I would have some more space to move the resistors away from J35 and J36.
[image]

Let us talk about pin assignment.
Also Wokwi simulation.

ForrestErickson, 2023-10-13:
Rotated and pushed around parts to make room.
[image]

[image]

LokiMetaSmith, 2023-10-16:
I believe we can close this @ForrestErickson  for now, for reference though, https://www.amazon.com/gp/product/B08P5LXR86/
this is the display board I'm working with.

ForrestErickson, 2023-10-17:
Regarding, ", https://www.amazon.com/gp/product/B08P5LXR86/
this is the display board I'm working with."  This should be added to the README.md where ever it is we are documenting our BOM.  I think we have way too scattered documentation and no index to it.

LokiMetaSmith, 2023-11-16:
@ForrestErickson is this resolved?