PubInv/NASA-COG #360: Mechanical 3D Model for Exact RepRap, Perhaps RepRap With Buzzer

ForrestErickson, 2023-10-19:
# Mismatch of RepRap and 3D Model

## The 3D Model
Lee pulled a 3D model from: https://grabcad.com/library/bigtreetech-tft24-v1-1-1

## The BIGTREETECH Mini12864 LCD Graphic Smart Display Control Board
https://www.amazon.com/gp/product/B08P5LXR86/ref=ox_sc_act_title_6?smid=A2K1WYI5NZ8QSW&psc=1

## Issues
This BIGTREETECH device does not appear to have a buzzer.

LokiMetaSmith, 2023-10-20:
@ForrestErickson 
take a look at the git repo for the board, it has the 3D model and more. 
https://github.com/bigtreetech/MINI-12864/tree/master/mini12864_v2.0/3D

the things in red are the beeper
[image]

ForrestErickson, 2023-10-21:
Added BigTreeTech display into TwoU assembly.
[image]
and
[image]

A note, In this TwoU configuration, mounted to the faceplate, the buzzer will not be heard much if at all.

ForrestErickson, 2023-10-21:
@RobertLRead 
@LokiMetaSmith 

Regarding the above, the Bigtreetech ?RepRap? assembly  has a buzzer that will unlikely be heard if mounted on a front panel  in a large enclosure. 

Do we care?  Do we have any requirement a buzzer fulfills?

I ask because I want to get any unstated assumptions resolved.

Lee

LokiMetaSmith, 2023-10-21:
It's not directly tied to any requirements, but as with the GPAD, an
audible alarm or indication of state change is useful.
The goal of the display module is to accelerate development as a known
working test platform, we may find that we want a different configuration
moving forward.