PubInv/NASA-COG #361: Need Due Sketch for Factory Testing RepRap Display Features

ForrestErickson, 2023-10-19:
We would benefit from a wokwi sketch (Arduino IDE) that can test the RepRap display on the Control V1.1 assembly. 
This would start as a factory test.
It might then become the basis of a new ?task? in the OEDCS code too.

[ ] Must write to the display a message
[ ] Must turn on back light (of appropriate)
[ ]Must react to rotary encoder so that the operator and the serial port interface can note success.
[ ] Must exercise a buzzer

LokiMetaSmith, 2023-10-19:
I have one started, but that was dropped when the display feature was removed and we committed to using the Stage2 firmware modification. 

I'll upload the partial sketch, it's just a gui display, rotary encoder project, and autoPID, but might be perfect for testing out those features on the board quickly

LokiMetaSmith, 2023-10-19:
NASA-COG\firmware-development\stage_2_heater\stage_2_heater.ino

libraries used 
https://github.com/adafruit/Adafruit-MAX31855-library
https://ryand.io/AutoPID/
https://github.com/mprograms/SimpleRotary
https://github.com/olikraus/u8g2

ForrestErickson, 2023-10-23:
@LokiMetaSmith 
Can you work up a test in Wokwi so that I can test the Control V1.1 assemblies when they arrive?
The assemblies are in (or the next step is in) JLCPCB packing now.

LokiMetaSmith, 2023-10-23:
The display is not one that has a Wokwi simulacrum.

Please don't delay sending boards out with that particular requirement.

Factory test should primarily verify pin orientation and expected
connection with the cables, the display and the schematic.
I recommend marking which cable end goes to which  board, as they are not
bidirectional.

There are several example programs for the display that should be
modifiable, the example program in the firmware-development

ForrestErickson, 2023-10-23:
@LokiMetaSmith 

Regarding, "Factory test should primarily verify pin orientation and expected"  The fastest way to do this will be functional test code.   Or please write up a test.

Regarding, "There are several example programs for the display that should be modifiable, the example program in the firmware-development"
Please provide an exact example.  

You have a display and a Due with which you can prepare and test code before I will have anything with which to work. This is critical path activity.

LokiMetaSmith, 2023-10-23:
Regarding "Regarding, "Factory test should primarily verify pin orientation
and expected" The fastest way to do this will be functional test code. Or
please write up a test."
Please confirm that this is a CRITICAL feature that overrides firmware
development and delivery for Task 2

Regarding "Regarding, "There are several example programs for the display
that should be modifiable, the example program in the firmware-development"
Please provide an exact example.

You have a display and a Due with which you can prepare and test code
before I will have anything with which to work. This is critical path
activity."
I can do this, but I would prefer to focus on that after the 27th as that
is when Task 2 is due.