PubInv/NASA-COG #362: For SD Card Access, Require RepRap Mounting Bezel

ForrestErickson, 2023-10-19:
The RepRap display has an SD card socket.

To access this the display will have to be mounted outside the Front Panel (what ever front panel). So a bezel that holds the display and mounts to the Front Panel will be required.

Something similar to this perhaps:
https://www.thingiverse.com/thing:3214103
[image]

Perahpes even editing this would work.  
It has License CC BY
[Reprap display - TEVO](https://www.thingiverse.com/thing:3214103)
by [raf_ds](https://www.thingiverse.com/raf_ds) is licensed under the [Creative Commons - Attribution](https://creativecommons.org/licenses/by/4.0/) license.

LokiMetaSmith, 2023-10-19:
I used this model 
https://www.printables.com/en/model/537527-articulating-mount-for-btt-fysetc-mini-12864-lcd-s

it fit well, printed easily and mounted to extruded aluminum

ForrestErickson, 2023-10-21:
Assembly as surface mounted to TwoU face plate

[image]

Note: The STEP comes in at a weird angle. 
[image]

ForrestErickson, 2023-10-21:
I should note that mounting the display like this, outside the enclosure, is a recipe for making EMC problems. This is how you would make a radiating / receiving antenna.

LokiMetaSmith, 2023-10-21:
[image]

https://www.makerbase.store/products/mks-mini12864-v3-insert-sd-card-side-lcd-smart-display-screen

There's a version with a front mounted SD card adapter. 

I would note this and apply it to what the final interface looks like

ForrestErickson, 2023-10-21:
Nice link.
The drawing helps too.
[image]

I note with disappointment they do not have dimensions to the center of the reset button.

Bigtreetech User Manual:
https://github.com/bigtreetech/MINI-12864/blob/master/mini12864_v2.0/BIGTREETECH%20Mini12864%20V2.0%20User%20Manual.pdf

Where I found the full dimensions:
[image]

LokiMetaSmith, 2023-10-23:
on the 3D printed model that I used, the reset pin needed to be modified to fit.