PubInv/NASA-COG #364: Does The OEDCS Need Air Flow Sensors To Capture Information Necessary For Thermal modeling of the System?

ForrestErickson, 2023-10-19:
Do we need to know the volume and therefor mass of incoming air and outgoing air to capture the heat flow of the system?

For the deliverable of the thermal model this is Lee's intuition.

RobertLRead, 2023-10-19:
It would be better. We could resurrect the MassAirflow system we used, which used 12V power, a voltage divider, and an Analog to Digital Converter to do it.

Or, we could calibrate a CFC with a flowmeter, and assume they are more or less the same (I'm skeptical of this.)

I think it would be better to have this data (and I think AmOx) has it. But I'm not sure of any sense in which it is absolutely necessary.

LokiMetaSmith, 2023-10-20:
I posted this in email, but this would be ideal to integrate into the airstream if we need to accurately meter the mass air flow. 

https://www.te.com/usa-en/products/sensors/mass-air-flow-sensors.html?tab=pgp-story

ForrestErickson, 2023-10-21:
I put this in an email  of 20231021 7:26 but realized it should be better captured in this issue or also captured in this issue.

Robert,
Lawrence,
 
During discussion with Lawrence this week I advanced my understanding of details what the oxygen generation system is physically. And this leads to how it might be modeled and my conclusion that we must have an air flow sensor.

Of use is the paper at: 
https://lpsa.swarthmore.edu/Systems/Thermal/SysThermalModel.html#:~:text=To%20model%20this%20system%20with,to%20the%20thermal%20system%20above
 
The last model is almost the oxygen generation system.
 
[image]

 
What I think needs to be added is the energy leaving the system in the form of hot O2. 
 
I was approximating the problem above in KiCAD as below with a Voltage to Current polynomial source. I had several days struggling to get the symbol to work in KiCAD as the KiCAD implementation was incomplete and requires some under the hood net naming in arguments to get it to function.
[image]

File in repo at: https://github.com/PubInv/NASA-COG/tree/develop/experimentation/KiCadThermo/RCRC/RCRC
 
However this model, as currently draw, does not yet model "Fluid flows into the tank at temperature θin with mass flow rate Gin, and specific heat cp". 
I was interrupted before I got to that and have not gotten back to it.
 
We must also know the temperature of the gas out of the heat exchanger.  I conclude my understanding of where the three thermocouples are in the system in Austin is insufficient to know if our OEDCS Serial Number 1 system can measure the temperature of the gas out. 
 
With out temperature of the gas out and the mass flow we have insufficient parameters to complete the modeling.

ForrestErickson, 2023-10-21:
@RobertLRead 

Can you add a thermocouple and an additional 1 Wire amplifier to the system at Austin in the near future?
I have no mental model of what the exhaust gas port looks like. Photos in a new private repository would help.

LokiMetaSmith, 2023-10-23:
The exhaust gas port is a slit of sheet metal and would be difficult to measure airflow, but temperature should be relatively trivial.

Note, AmOx used a long rigid tube to act as a laminar flow air guide to then use a hand held air flow meter such as https://www.amazon.com/Digital-Anemometer-Handheld-Accurately-Temperature/dp/B07J64TCBQ/

As the air flow resistance doesn't change, measuring the state of the system once is sufficient for calculations as long as temperature is known