PubInv/NASA-COG #366: Any Failsafe Must Work With long Start Time of Heartbeat Signal

ForrestErickson, 2023-10-19:
# Design of Failsafe must be compatible with start up.

Rev: 0.3.8 code adds a Failsafe function toggling the LED_BUILTIN task. 
The Heartbeat task does not start till quite some time after the Due is reset.

## Measured Delay of Heartbeat
Measure from reset to first toggle of heartbeat signal (on LED_BUILTIN).
[image]
CH3 is Reset, CH4 is D13 (LED_BUILTIN)

## Delay Time
Almost 24 seconds delay.

RobertLRead, 2023-10-19:
Note, however, that the initialization code could easily toggle the heartbeat manually during the initialization of the hardware. This could lower this to 5 seconds pretty easily, and may be a better solution. You can call the "_run" function on a task before the main processor loop starts, which means you would only have to put the actual toggle code in one place (that task.)

LokiMetaSmith, 2023-10-20:
Likely the delay is also exasperated as we added the code to the very end of the task initialization routine, of which many time out due to not having devices present.

ForrestErickson, 2023-10-20:
Regarding "call the "_run" function on a task before the main processor loop starts" and "added the code to the very end of the task initialization routine"
Both of which are telling us we should characterize all of the initialization times just as we should characterize all of the task run times for a worst case.  I assume there is some tedious and hardware dependent things to learn.

I have also speculated that the failsafe may have to be enabled by the controller or it may itself have to have a longish enable time from power up.  It feels scary contemplating what might go wrong and has me wondering how to even google for good ideas (which I have not yet done).

ForrestErickson, 2023-10-20:
More thoughts. If our serial port reporting can be made more verbose (to find the worst case it can cause) we should do so ASAP.

@RobertLRead  Can and if so how to make it more verbose?

RobertLRead, 2023-10-20:
Yes. I usually use DEBUG_LEVEL flags to control verbosity, and I would
recommend that here.

The basic approaches that we build up a structure which is a "report" and
then report it in a "pretty" form on the serial port in a task. The
"pretty" form can be expanded, and more information can easily be added to
the report.

That is not great for "events" though, which happen at a moment in time.

Does that help?

ForrestErickson, 2023-10-20:
@RobertLRead 
Regarding, "I usually use DEBUG_LEVEL flags to control verbosity, and I would
recommend that here."  I am asking how to turn on the existing reporting to the maximum level.  I may already have been shown once but once is not enough for my old brain I admit. 

Notes here would help.

RobertLRead, 2023-10-20:
I'm sorry I am travelling and we can't have a quick conversation about this.

I use the DEBUG_LEVEL paradigm for actual debugging statements. I am
suggesting we create "REPORT_VERBOSITY_LEVEL" and allow the report to
become more verbose based on this level.

The file /firmware/lib/task/machine.cpp has a function "outputReport" which
outputs to the serial report. If you add a new field to the
"MachineStatueReport" structure, you can print it here. You could control
whether or not it gets printed with a "REPORT_VERBOSITY_LEVEL" integer.

If you want to print it out more often, you can change its period:

serialReportTask.PERIOD_MS

(but this is in the class definition for that task.)

I'm not sure I've answered your question.