PubInv/NASA-COG #379: Write, Factory Test Plan,  Control V1.1, Printed Wiring Assembly

ForrestErickson, 2023-10-25:
Bench testing and functional testing in a markdown document.

ForrestErickson, 2023-10-25:
# Factory Test Plan, Control V1.1, Printed Wiring Assembly

Summary and data capture table

| Test ID 	| Test Name       	| Setup                                          	| Test Location 	| Requirements    	| Results 	| Notes 	|
|---------	|-----------------	|------------------------------------------------	|---------------	|-----------------	|---------	|-------	|
| 0       	| Inspection      	| Inspect correct orientation on polarized parts 	|               	|                 	|         	|       	|
| 1       	| +24 In No Short 	| Set ohm meter 20M. Com to GND                  	| J12 Pin 1     	| Open or > 1 Meg 	|         	|       	|
| 2       	| +12 In No Short 	| Set ohm meter 20M. Com to GND                  	| J11 Pin 1     	| Open or > 1 Meg 	|         	|       	|
| 3       	| 5VHEAD net      	| Set ohm meter 20M. Com to GND                  	| J8 Pin 1      	| Open or > 1 Meg 	|         	|       	|
| 4       	| +5V net         	| Set ohm meter 20M. Com to GND                  	| J3 Pin 4      	| Open or > 1 Meg 	|         	|       	|
| 5       	| +3.3V net       	| Set ohm meter 20M. Com to GND                  	| J3 Pin 4      	| Open or > 1 Meg 	|         	|       	|
| 6       	|                 	|                                                	|               	|                 	|         	|       	|
| 7       	|                 	|                                                	|               	|                 	|         	|       	|
| 8       	|                 	|                                                	|               	|                 	|         	|       	|
|         	|                 	|                                                	|               	|                 	|         	|       	|

ForrestErickson, 2023-10-25:
Re: "markdown file somewhere?"  Not yet. 

I am experimenting and not happy with what I am coming up with.  

I also googled about opensource test management tools. Example:  "online free software and hardware test management tools Github " but all I am finding involves more reading that the rewards justify.

The ideal thing would let us add records. Edit records as the common database features would have. 

I have no experience with on line databases but have to thinks something must exist.   
Just thinking out loud, Is this a possible use case for a google doc even?

ForrestErickson, 2023-10-27:
# Factory Test Plan, Control V1.1, Printed Wiring Assembly
Capture of test of the build of October 2023.
Units sterilized as 9-13.

## Unpowered Test of PWA.
Test equipment: Multimeter (EMCO DMR-3800),
Summary and data capture table

Tester Name _____________________________________
Date of Test ______________________________________

| Test ID 	| Test Name       	| Setup                                          	| Test Location 	| Requirements    	| Results SN 9 	| Results SN 10 	| Results SN 11 	| Results SN 12 	| Results SN 13 	|
|--------:	|-----------------	|------------------------------------------------	|---------------	|-----------------	|--------------	|---------------	|---------------	|---------------	|---------------	|
|       1 	| Inspection      	| Inspect correct orientation on polarized parts 	|               	|                 	|              	|               	|               	|               	|               	|
|       2 	| +24 In No Short 	| Set ohm meter 20M. Com to GND                  	| J12 Pin 1     	| Open or > 1 Meg 	|              	|               	|               	|               	|               	|
|       3 	| +12 In No Short 	| Set ohm meter 20M. Com to GND                  	| J11 Pin 1     	| Open or > 1 Meg 	|              	|               	|               	|               	|               	|
|       4 	| 5VHEAD net      	| Set ohm meter 20M. Com to GND                  	| J8 Pin 1      	| Open or > 1 Meg 	|              	|               	|               	|               	|               	|
|       5 	| +5V net         	| Set ohm meter 20M. Com to GND                  	| J3 Pin 4      	| Open or > 1 Meg 	|              	|               	|               	|               	|               	|
|       6 	| +3.3V net       	| Set ohm meter 20M. Com to GND                  	| J3 Pin 4      	| Open or > 1 Meg 	|              	|               	|               	|               	|               	|
|       7 	|                 	|                                                	|               	|                 	|              	|               	|               	|               	|               	|
|       8 	|                 	|                                                	|               	|                 	|              	|               	|               	|               	|               	|
|       9 	|                 	|                                                	|               	|                 	|              	|               	|               	|               	|               	|
|         	|                 	|                                                	|               	|                 	|              	|               	|               	|               	|               	|

LokiMetaSmith, 2023-10-27:
i would add a section to talk about the pin polarity of the display, but I would also test the display after testing the main board

ForrestErickson, 2023-11-09:
## 20231109 Resume testing on SN9.
The sketch MAX31850_Tester;_Rev_0.2 has been running since my last work continuously. 

## Sneak Up on Testing Ethernet with Control V1.1
*  Connect the DUE to the Ethernet WITH OUT the Control V1.1.
Update local repository clone and load firmware. 
Set for "-D ALLOW_BAD_THERMOCOUPLES_FOR_TESTING". Compile and load.
It boots and reports version "Starting OEDCS; Rev: 0.3.10"
DUT got network time.

*  Connect Due to Control V1.1 assembly.
Apply power and connect with Terminal
Normal boot messages and the expected ethernet failure

> 12:35:30.744 -> W5x00 init failed
> 12:35:31.354 -> W5x00 init failed
> 12:35:32.010 -> W5x00 init failed
> 12:35:32.621 -> W5x00 init failed
> 12:35:33.234 -> W5x00 init failed
> 12:35:33.842 -> W5x00 init failed
> 12:35:34.450 -> W5x00 init failed
> 12:35:35.106 -> W5x00 init failed
> 12:35:35.714 -> W5x00 init failed
> 12:35:36.323 -> W5x00 init failed
> 12:35:36.323 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
> 12:35:41.344 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
> 12:35:46.350 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!

* Connect the Ethernet Sheild
Apply power. Connect USB port. Open Terminal window
the DUT found the thermocuples and continued but did not find the Ethernet sheild

> 12:40:10.379 -> CogTask init
> 12:40:10.379 -> OEDSCSerialTask Inited
> 12:40:10.998 -> W5x00 init failed
> 12:40:11.604 -> W5x00 init failed
> 12:40:12.238 -> W5x00 init failed
> 12:40:12.828 -> W5x00 init failed
> 12:40:13.483 -> W5x00 init failed
> 12:40:14.080 -> W5x00 init failed
> 12:40:14.731 -> W5x00 init failed
> 12:40:15.333 -> W5x00 init failed
> 12:40:15.954 -> W5x00 init failed
> 12:40:16.590 -> W5x00 init failed
> 12:40:16.590 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
> 12:40:21.571 -> CRITICAL ERROR! CONFIGURED FOR ETHERNET, BUT NONE FOUND!
 

Moving Ethernet troubleshooting to issue #261