PubInv/NASA-COG #380: watchdog timer never trips if Ethernet plug isn't inserted, or

LokiMetaSmith, 2023-10-25:
when plugged into the network, the device will run as expected with a working ethernet card, however with nothing plugged in, it will trigger the watchdog timer and reset the system. 

when an ethernet shield or ethernet cable or network connection are not present, the device will run  and halt with "CRITICAL ERROR, CONFIGURED FOR ETHERNET BUT NONE FOUND!"

need to confirm intended behavior

RobertLRead, 2023-10-25:
> Assigned #380 <https://github.com/PubInv/NASA-COG/issues/380> to
> @RobertLRead <https://github.com/RobertLRead>.

The critical error is intended behavior.

I don't know what we should do if the power supply is not found. We should
not let the watchdog trip because of not having hardware---we should report
it instead.

ForrestErickson, 2023-10-26:
There are multiple customers here. 

1. The end user of the final product.
2. There will be troubleshooting personnel who may need to have a way to probe the system and that means lock out some safety features. Safety features must return at least after a power cycle when a device might power up unattended.
3. And use the developers who may need to operate in partial systems.  For developers a compile time way to disable features is important.  How do we ensure they do not get released into the wild?