PubInv/NASA-COG #387: Control PCB Improvements List, Towards Version 1.2, Hardware

ForrestErickson, 2023-10-30:
Notes on improvements for a version 1.2 PCB if ever there is to be one

-[ ] Add Test point for 5VHEAD net
-[ ] Add test point for +5V net
-[ ] Add test point for +3.3V net
-[ ] Connect J3 pin 8 to the Due

ForrestErickson, 2023-10-30:
@LokiMetaSmith  Can you help me with the markdown syntax for check lists in the above? I believe I am following the instructions from: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/about-task-lists but not getting the expected result.

LokiMetaSmith, 2023-10-30:
> @LokiMetaSmith Can you help me with the markdown syntax for check lists in the above? I believe I am following the instructions from: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/about-task-lists but not getting the expected result.

you were missing a space between "-[" it should have been "- ["

ForrestErickson, 2023-10-30:
Dang white spaces....