PubInv/NASA-COG #393: Heater duty cycle is reporting 100% when SSR is not steady on

tomsrobots, 2023-11-15:
During heatup of the mule, I noticed the OEDCS reporting "Heater DC: 1.000" with the setpoint far ahead of the heater temperature.

```
Machine State: Warmup
Target     C: 600.00
Setpoint   C: 519.00
Target Ramp C: 0.50
Max Stack   A: 0.00
Max Stack   W: 100.00
Fan PWM      : 1.00
Post Heater C: 428.00
Post Getter C: 433.75
Post Stack  C: 477.00
Heater DC    : 1.00000
Stack amps  A: 0.69
Stack watts W: 5.42
Stack volts V: 7.85
Stack ohms  O: 11.38
Fan RPM      : 2768.00
Enter s:1 to Turn On, s:0 to Turn Off.
Enter a:XX.X to set (a)mperage, (w)attage, (f)an speed (h)eater set p., and (r)amp rate.
Putting out speed to fan control board:0
PWM ratio:  num / ratio : 1.00
RPMS:
rpm: 2768
```

Concerned the controller wasn't giving enough to the unit, I went out to the floor and checked the SSR. I didn't time it, but the indicator light looked like it might be on 70% of the time. I verified this on/off nature with the multimeter. There is some issue translating the expected duty cycle variable to the real on/off time of the SSR.

LokiMetaSmith, 2023-11-15:
I suspect this issue is related to #388

RobertLRead, 2023-11-15:
This is an important bug. I'm starting to work on it.

RobertLRead, 2023-11-17:
This has been fixed. It has been deployed and Tom notes no problem; I do not believe he has fully tested it. Nonteheless I am closing this issue assuming it to be fixed.

closed 2023-11-17 by RobertLRead